muskie82/MonoGS #1: CUBLAS_STATUS_NOT_SUPPORTED

kshitijgoel007, 2024-03-04:
Running
```
python slam.py --config configs/mono/tum/fr3_office.yaml
```
results in the error
```
Traceback (most recent call last):
  File "/home/fractal/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/fractal/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/fractal/workspaces/MonoGS/utils/slam_backend.py", line 407, in run
    self.initialize_map(cur_frame_idx, viewpoint)
  File "/home/fractal/workspaces/MonoGS/utils/slam_backend.py", line 90, in initialize_map
    viewpoint, self.gaussians, self.pipeline_params, self.background
  File "/home/fractal/workspaces/MonoGS/gaussian_splatting/gaussian_renderer/__init__.py", line 66, in render
    projmatrix=viewpoint_camera.full_proj_transform,
  File "/home/fractal/workspaces/MonoGS/utils/camera_utils.py", line 102, in full_proj_transform
    self.projection_matrix.unsqueeze(0)
RuntimeError: CUDA error: CUBLAS_STATUS_NOT_SUPPORTED when calling `cublasSgemmStridedBatched( handle, opa, opb, m, n, k, &alpha, a, lda, stridea, b, ldb, strideb, &beta, c, ldc, stridec, num_batches)`
```

The conda environment is configured exactly as prescribed in your README.

rmurai0610, 2024-03-04:
Hi, thank you for your interest in our project.

This seems to be a common issue:
https://github.com/graphdeco-inria/gaussian-splatting/issues/184
and for Ubuntu 18.04 we had to downgrade the cuda-toolkit to 11.3.
(whereas for 20.04, we had the opposite issue where we must use 11.6)

If you can try manually reducing the cuda-toolkit version in the [environment.yml](https://github.com/muskie82/MonoGS/blob/main/environment.yml#L7) that would be great and if anyone knows a better way to detect/fix this, please let us know!

chengYi-xun, 2024-03-06:
@rmurai0610 Thank you very much for your work on this, I have resolved the environmental issue through the following actions.

`my computer:cuda-11.8 ubutn 18.04LTS
###############################
create conda env:
conda create -n monogs python=3.7.13
conda activate monogs  
######################
solve cuda-toolkit(my computer use 1 hours to run this):
conda install -c "nvidia/label/cuda-11.6.0" cuda-toolkit
####################
install torch:
conda install pytorch==1.12.1 torchvision==0.13.1 torchaudio==0.12.1 cudatoolkit=11.6 -c pytorch -c conda-forge
#####################
pip install -r req.txt 
cd submodules/diff-gaussian-rasterization/
python setup.py install
pip install .
cd simple-knn/
python setup.py install
pip install .

req.txt:

opencv-python==4.8.1.78
munch
trimesh
evo==1.11.0
open3d==0.17.0
torchmetrics
imgviz
PyOpenGL
glfw
PyGLM
wandb
lpips
rich
ruff
plyfile==0.8.1
tqdm

`

closed 2024-03-12 by rmurai0610