muskie82/MonoGS #10: Test on custom data

leblond14u, 2024-03-06:
Hi thanks again for the amazing work,

Is there a way to test the SLAM system on custom datasets ? 
Can you indicate me the procedure to do so ? 

Thanks in advance,
Best regards,

Hugo

muskie82, 2024-03-06:
Hi,

Check out dataset/parser class implemented in [dataset.py](https://github.com/muskie82/MonoGS/blob/main/utils/dataset.py).
Prepare your custom dataset/parser class, then add it to `load_dataset` function.

closed 2024-03-14 by rmurai0610

hnglp, 2024-08-30:
If I don't know the ground truth or RGB values of my dataset and don't have files like "rgb.txt," I can't create this parser i guess. What would you recommend in this case? I'm having difficulty finding datasets that include this information apart from TUM. @muskie82