muskie82/MonoGS #4: ImportError: /lib64/libm.so.6: version `GLIBC_2.29' not found

WANGYINGYU, 2024-03-06:
Hi,

When I try to run the demo using

`python slam.py --config configs/mono/tum/fr3_office.yaml
`
, the program outputs the following error

```
Traceback (most recent call last):
  File "slam.py", line 15, in <module>
    from gui import gui_utils, slam_gui
  File "/home/yingywan/Documents/MonoGS/gui/slam_gui.py", line 19, in <module>
    from gui.gl_render import util, util_gau
  File "/home/yingywan/Documents/MonoGS/gui/gl_render/__init__.py", line 1, in <module>
    from . import render_ogl, util, util_gau
  File "/home/yingywan/Documents/MonoGS/gui/gl_render/render_ogl.py", line 7, in <module>
    from . import util, util_gau
  File "/home/yingywan/Documents/MonoGS/gui/gl_render/util.py", line 1, in <module>
    import glm

ImportError: /lib64/libm.so.6: version `GLIBC_2.29' not found (required by /home/xxx/miniconda3/envs/MonoGS/lib/python3.7/site-packages/glm.cpython-37m-x86_64-linux-gnu.so)
```

I think it is caused by my Linux system version, which is the version of GLIBC is 2.28, but I can't install GLIBC_2.29 by myself since I use the university's cluster. 
Do you know how to solve this problem?

muskie82, 2024-03-06:
Hi,

Can you try building and installing PyGLM locally by cloning the repo?
https://github.com/Zuzu-Typ/PyGLM/tree/master

Another way is just delete all the gl_render related code in slam_gui.py.
Since gl_render package is used just for Elipsoid shader, the demo works without it.

WANGYINGYU, 2024-03-06:
> Hi,
> 
> Can you try building and installing PyGLM locally by cloning the repo? https://github.com/Zuzu-Typ/PyGLM/tree/master
> 
> Another way is just delete all the gl_render related code in slam_gui.py. Since gl_render package is used just for Elipsoid shader, the demo works without it.

Thank you for your reply. I solved this problem by installing PyGLM with the version 2.7.0. The reason for this problem is the latest version (2.7.1) removed the support of python 3.7.

closed 2024-03-06 by muskie82