muskie82/MonoGS #9: GaussianRasterizationSettings.__new__() got an unexpected keyword argument 'projmatrix_raw'

leblond14u, 2024-03-06:
Hi thanks for your great work on Gaussian splatting SLAM. 

I am encountering an issue while trying to test your implementation : 
`MonoGS: saving results in result/datasets_tum/2024-03-06-13-49-41
FEngine (64 bits) created at 0x56021834d430 (threading is enabled)
FEngine resolved backend: OpenGL
MonoGS: Resetting the system
Process Process-3:
Traceback (most recent call last):
  File "/home/hleblond/anaconda3/envs/MonoGS/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/hleblond/anaconda3/envs/MonoGS/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/hleblond/Bureau/PhD/MonoGS/utils/slam_backend.py", line 407, in run
    self.initialize_map(cur_frame_idx, viewpoint)
  File "/home/hleblond/Bureau/PhD/MonoGS/utils/slam_backend.py", line 89, in initialize_map
    render_pkg = render(
                 ^^^^^^^
  File "/home/hleblond/Bureau/PhD/MonoGS/gaussian_splatting/gaussian_renderer/__init__.py", line 58, in render
    raster_settings = GaussianRasterizationSettings(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: GaussianRasterizationSettings.__new__() got an unexpected keyword argument 'projmatrix_raw'` 

Did someone already encounter this issue ?

Thanks in advance,
Best regards,

Hugo

muskie82, 2024-03-06:
Hi, thanks for your interest in our work.

Could you double check you installed [our version of diff-rasterization](https://github.com/rmurai0610/diff-gaussian-rasterization-w-pose)?
It happened to me when I used different one.

leblond14u, 2024-03-06:
My bad ! 
Indeed I had the wrong version installed. Works with this one [diff-rasterization](https://github.com/rmurai0610/diff-gaussian-rasterization-w-pose).
Thanks a lot :)

closed 2024-03-06 by leblond14u