muskie82/MonoGS #15: Gaussian Pruning for Monocular case

8pterosaur, 2024-03-09:
It's a very good job and the code is very elegant

### Q1: In the paper "Guassian Insertion and Pruning" get the second paragraph 
> If the Gaussians inserted within the last 3 keyframes are unobserved by at least 3 other frames, we prune them out as they are geometrically unstable.

In the map update of backend, the part code:

```
if prune:
  if len(current_window) == self.config["Training"]["window_size"]:
      prune_mode = self.config["Training"]["prune_mode"]
      prune_coviz = 3
      self.gaussians.n_obs.fill_(0)
      for window_idx, visibility in self.occ_aware_visibility.items():
          self.gaussians.n_obs += visibility.cpu()
      to_prune = None
      if prune_mode == "odometry":
          to_prune = self.gaussians.n_obs < 3
          # make sure we don't split the gaussians, break here.
      if prune_mode == "slam":
          # only prune keyframes which are relatively new
          sorted_window = sorted(current_window, reverse=True)
          mask = self.gaussians.unique_kfIDs >= sorted_window[2]
          if not self.initialized:
              mask = self.gaussians.unique_kfIDs >= 0
          to_prune = torch.logical_and(
              self.gaussians.n_obs <= prune_coviz, mask
          )
      if to_prune is not None and self.monocular:
          self.gaussians.prune_points(to_prune.cuda())
          for idx in range((len(current_window))):
              current_idx = current_window[idx]
              self.occ_aware_visibility[current_idx] = (
                  self.occ_aware_visibility[current_idx][~to_prune]
              )
      if not self.initialized:
          self.initialized = True
          Log("Initialized SLAM")
      # # make sure we don't split the gaussians, break here.
  return False
```
According to my understanding, the `sort_window[2]` means **last 3 keyframes** in the paper, the `self.gaussians.unique_kfIDs` means all gaussians unique id in the map. So, how should I understand **by at least 3 other frames** in the paper.  Isn’t it the remaining frames in the current window except for the three newly inserted key frames? 

### Q2: It's also related to 'self.gaussians.unique_kfIDs'
According to the every time unique_kfIDs updates in extend_from_pcd() :
```
new_unique_kfIDs = torch.ones((new_xyz.shape[0])).int() * kf_id
```
It is bound to the ID of the current keyframe. 
prune_points
densify_and_split
densification_postfix

rmurai0610, 2024-03-09:
Hi, thank you for your interest!
q1/ Yes, you are right, this part of the code finds all the Gaussians which were created within the last 3 keyframes
```
sorted_window = sorted(current_window, reverse=True)
mask = self.gaussians.unique_kfIDs >= sorted_window[2]
```
3 other frames mean that if a Gaussian is inserted by frame $i$, it should be observed by 3 other frames $j, k, l$ (i.e. 4 frames in total), and $j, k, l$ are any frames in the keyframe window apart from $i$.
We didn't mean that the Gaussians must be observed from the three frames excluding the latest 3 frames.
Such a check is done here
https://github.com/muskie82/MonoGS/blob/main/utils/slam_backend.py#L261-L263
I hope this helps clear up the confusion.

q2/  Yes, you are right, Gaussians are assigned the ID of the keyframes which created the Gaussians and are used for pruning.

8pterosaur, 2024-03-10:
great, thx.

closed 2024-03-10 by 8pterosaur