muskie82/MonoGS #16: Build diff-gaussian-rasterization failed.

wongshek, 2024-03-12:
- OS: Windows 10
- CUDA outside: CUDA 11.8
- CUDA inside: CUDA 11.6

```
Processing f:\test\monogs\submodules\diff-gaussian-rasterization
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: diff_gaussian_rasterization
  Building wheel for diff_gaussian_rasterization (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [62 lines of output]
      running bdist_wheel
      D:\Programs\miniconda3\envs\monogs\lib\site-packages\torch\utils\cpp_extension.py:411: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      running build
      running build_py
      copying diff_gaussian_rasterization\__init__.py -> build\lib.win-amd64-cpython-37\diff_gaussian_rasterization
      running build_ext
      D:\Programs\miniconda3\envs\monogs\lib\site-packages\torch\utils\cpp_extension.py:813: UserWarning: The detected CUDA version (11.8) has a minor version mismatch with the version that was used to compile PyTorch (11.6). Most likely this shouldn't be a problem.
        warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
      building 'diff_gaussian_rasterization._C' extension
      "C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.8\bin\nvcc" -c cuda_rasterizer/backward.cu -o build\temp.win-amd64-cpython-37\Release\cuda_rasterizer/backward.obj -ID:\Programs\miniconda3\envs\monogs\lib\site-packages\torch\include -ID:\Programs\miniconda3\envs\monogs\lib\site-packages\torch\include\torch\csrc\api\include -ID:\Programs\miniconda3\envs\monogs\lib\site-packages\torch\include\TH -ID:\Programs\miniconda3\envs\monogs\lib\site-packages\torch\include\THC "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.8\include" -ID:\Programs\miniconda3\envs\monogs\include -ID:\Programs\miniconda3\envs\monogs\Include "-ID:\Programs\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-ID:\Programs\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.2\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" -Xcudafe --diag_suppress=dll_interface_conflict_dllexport_assumed -Xcudafe --diag_suppress=dll_interface_conflict_none_assumed -Xcudafe --diag_suppress=field_without_dll_interface -Xcudafe --diag_suppress=base_class_has_different_dll_interface -Xcompiler /EHsc -Xcompiler /wd4190 -Xcompiler /wd4018 -Xcompiler /wd4275 -Xcompiler /wd4267 -Xcompiler /wd4244 -Xcompiler /wd4251 -Xcompiler /wd4819 -Xcompiler /MD -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr -IF:\Test\MonoGS\submodules\diff-gaussian-rasterization\third_party/glm/ -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 --use-local-env
      F:\Test\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\auxiliary.h(151): warning #177-D: variable "p_proj" was declared 
but never referenced
     
      C:/Program Files (x86)/Windows Kits/10/include/10.0.19041.0/um\winnt.h(22866): warning #174-D: expression has no effect
     
      C:/Program Files (x86)/Windows Kits/10/include/10.0.19041.0/um\winuser.h(14630): warning #108-D: signed bit field of length 1
     
      C:/Program Files (x86)/Windows Kits/10/include/10.0.19041.0/um\winuser.h(14631): warning #108-D: signed bit field of length 1
     
      C:/Program Files (x86)/Windows Kits/10/include/10.0.19041.0/um\wincrypt.h(21791): warning #1835-D: attribute "dllimport" does not apply here
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(719): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(720): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(721): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(722): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(723): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(724): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(725): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(726): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(727): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(728): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(729): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(730): warning #108-D: signed bit field of length 1
     
      C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared\rpcndr.h(731): warning #108-D: signed bit field of length 1
     
      F:\Test\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(471): warning #177-D: variable "c" was declared but never referenced
                detected during instantiation of "void preprocessCUDA<C>(int, int, int, const float3 *, const int *, const float *, const __nv_bool *, const glm::vec3 *, const glm::vec4 *, float, const float *, const float *, const float *, const glm::vec3 *, const float3 *, glm::vec3 *, float *, float *, float *, float *, glm::vec3 *, glm::vec4 *, float *) [with C=3]"
      (838): here
     
      F:\Test\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(472): warning #177-D: variable "d" was declared but never referenced
                detected during instantiation of "void preprocessCUDA<C>(int, int, int, const float3 *, const int *, const float *, const __nv_bool *, const glm::vec3 *, const glm::vec4 *, float, const float *, const float *, const float *, const glm::vec3 *, const float3 *, glm::vec3 *, float *, float *, float *, float *, glm::vec3 *, glm::vec4 *, float *) [with C=3]"
      (838): here
     
      F:\Test\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(554): error: expected an expression
                detected during instantiation of "void renderCUDA<C>(const uint2 *, const uint32_t *, int, int, const float *, const float2 *, const float4 *, const float *, const float *, const float *, const uint32_t *, const float *, const float *, float3 *, float4 *, float *, float *, float *) [with C=3U]"
      (882): here
     
      1 error detected in the compilation of "cuda_rasterizer/backward.cu".
      backward.cu
      error: command 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v11.8\\bin\\nvcc.exe' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff_gaussian_rasterization
  Running setup.py clean for diff_gaussian_rasterization
Failed to build diff_gaussian_rasterization
Installing collected packages: diff_gaussian_rasterization
  Running setup.py install for diff_gaussian_rasterization ... error
  error: subprocess-exited-with-error
```

wongshek, 2024-03-12:
The problem has been resolved after add c++17 support in diff-gaussian-rasterization/setup.py

replace line29 with `extra_compile_args={"nvcc": ["-I" + os.path.join(os.path.dirname(os.path.abspath(__file__)), "third_party/glm/"),"--std=c++17"]})`

closed 2024-03-12 by wongshek

foreverlong, 2024-03-27:
Thank you for this solution, It woks. But do you successfully run this project on Windows? Have you ever meet this error:
```
torch._C._LinAlgError: torch.linalg.inv: The diagonal element 1 is zero, the inversion could not be completed because the input matrix is singular.
[W C:\cb\pytorch_1000000000000\work\torch\csrc\CudaIPCTypes.cpp:15] Producer process has been terminated before all shared CUDA tensors released. See Note [Sharing CUDA tensors]
```
Looking forward for your answer! Thank you!

Duuuuu, 2024-06-12:
> Thank you for this solution, It woks. But do you successfully run this project on Windows? Have you ever meet this error:
>
> ```
> torch._C._LinAlgError: torch.linalg.inv: The diagonal element 1 is [...]

I have meet the same question with you, have you solved this problem? Looking forward for you answer

ZW092719, 2025-04-15:
> I have meet the same question with you, have you solved this problem? Looking forward for you answer

I meet this too and the visualization seems error，hava you solve it？