muskie82/MonoGS #19: Question about ground truth pose use with 3DGS

leblond14u, 2024-03-12:
Dear authors,

Thank you for your amazing work ! 

I would like to know how you managed to use the GT poses with 3DGS. 
I am myself trying to benchmark the [ICL NUIM](http://redwood-data.org/indoor/dataset.html) dataset with the base 3DGS implementation using the ground truth poses. 
I wrote a custom data loader to replace the COLMAP estimated R, T and intrinsics with the ground truth ones for each camera objects. I also took note of the C2W to W2C transformation between the COLMAP norm and my dataset.
So to recap, I'm training the 3DGS with GT poses and intrinsics, GT colored point-cloud for initialization.

However I can't seem to get any result in doing so (see the screenshot attached). My camera poses seems off even though they are the GT poses. 
[image]

There is something I am clearly missing but can't seem to figure out what it is. 
Have you faced similar problems and/or do you know how to solve this issue ? 

Thanks a lot,

Hugo

muskie82, 2024-03-12:
Hi,

I am not sure about ICL-NUIM's dataset format.
Ground truth pose data might be in C2W or W2C, opengl-style or opencv-style, depending on the dataset... so it is not always straightforwad to try new dataset.

What I will do is
- Run COLMAP on ICL-NUIM, and train 3DGS with the COLMAP result. Confirm it actually works.
- Check the difference between the COLMAP pose and GT pose.

Best,

leblond14u, 2024-03-13:
Indeed what I'm doing right now as a workaround is to compute the COLMAP path for the dataset, then find the transformation between the colmap_C2W path and the GT_C2W path and then apply this transformation to the "colmap_W2C" to get a similar path in my GT scene. 
Still, not successful in having a correct reconstruction ...
So I thought maybe there was a more simple/systematic way of exploiting the ground truth paths for new datasets. 

Thanks for your answer,
Best regards,

closed 2024-04-01 by muskie82