muskie82/MonoGS #24: Question relates to Point Clound Generation

Oliverbihop, 2024-03-17:
This is such nice work!!!! 
I am a newbie in GS splatting. I have several questions.
1) As I read in the original paper of GS, they derived the Point Cloud (x,y,z) from SfM and stored it in COLMAP format.  In MonoGS, How is Point Cloud generated (I think it is from photometric difference right ? ). Where is the part of the code?
2) I read the code and guessed that the SLAM front-end and back-end run in parallel right? Can I have a short pseudo code from inputs to outputs for better understanding of codes?

Many thanks.

muskie82, 2024-03-18:
Thank you for your interest in our work.

1. Our method jointly estimates camera pose and 3D Gaussians from randomly initialised point cloud. You can find the relevant code [here](https://github.com/muskie82/MonoGS/blob/37cebb4b8df24777980eac00276c07059b94762c/gaussian_splatting/scene/gaussian_model.py#L122). Please check the detail in the paper.

2. Our method follows the standard tracking/mapping SLAM architecture, which runs time-critical camera tracking in the frontend while performs extensive 3D Gaussian mapping in the backend. Please check Fig.2 of [our paper](https://www.imperial.ac.uk/media/imperial-college/research-centres-and-groups/dyson-robotics-lab/hide-et-al_GaussianSplattingSLAM_Dec2023.pdf), and let me know if you have any specific part of the code you want to know more.

Thanks

Oliverbihop, 2024-03-19:
Sorry for my late reply. 
I got it
Thank you very much.

closed 2024-03-19 by muskie82