muskie82/MonoGS #26: Installation Error with simple_knn and diff_gaussian_rasterization

hnglp, 2024-03-18:
Hello,

Thanks for the great work. When I try to create the project env using Win 11 and Cuda 11.6 (added to the path, I checked it from the terminal and I updated the pytroch etc. version accordingly), I encounter an error while trying to install the packages named simple_knn and diff_gaussian_rasterization. This error seems to occur during compilation of packages. While compiling the simple_knn package, I get an error saying "too few arguments for template template parameter" when using the nvcc tool. Similarly, I get an error when using nvcc when compiling the diff_gaussian_rasterization package.

If you want, I can share the detailed output of the errors. Since it is a very long error text, I did not want to throw it away directly. What could be the cause of this problem?

Thanks.

muskie82, 2024-03-18:
Hi,

Thank you for your interest in our work.
Although we didn't test our code on Windows, I guess the error message is related to several issues reported in 3D Gaussian Splatting repo.
- https://github.com/graphdeco-inria/gaussian-splatting/issues/301
- https://github.com/graphdeco-inria/gaussian-splatting/issues/546
- https://github.com/graphdeco-inria/gaussian-splatting/issues/297
- https://github.com/graphdeco-inria/gaussian-splatting/issues/142

Can you check these issues/solutions?

Thanks,

hnglp, 2024-03-18:
Thank you for the quick response. After checking, I will share the current status here.

foreverlong, 2024-03-22:
I met this error too, have you finish it? Please tell me. Thank you!

hnglp, 2024-03-26:
OS: Win 11

I have updated the versions in the dependencies as follows:

```
dependencies:
  - cudatoolkit=11.7
  - plyfile=0.8.1
  - python=3.10.14
  - pip=22.3.1
  - pytorch=1.13.1
  - torchaudio=0.13.1
  - torchvision=0.14.1
  - tqdm
```
After downloading and installing CUDA from the NVIDIA website, you must add it to the PATH.

Then, I created the environment with conda create. I had to do this again because I noticed that the C/C++ compiler wasn't included by default in VS Code, and Ninja wasn't installed either. I added Ninja with Conda Forge and then added compilers from the extensions. Finally, I replaced line 29 in the setup.py file of the diff submodule with the code found in this [answer.](https://github.com/muskie82/MonoGS/issues/16#issuecomment-1989798377)

This way, I managed to set up the environment properly. If you encounter any errors after trying these steps, feel free to let me know, and I'll try to help after ensuring I haven't missed anything during my attempts.

After successfully setting up the environment, I immediately posted this response, hoping that the version changes I made due to compatibility issues wouldn't cause any problems during the project's execution :)

foreverlong, 2024-04-01:
> response

Thank you for your response! I apologize for the late reply as I was sick for a week. I have successfully built this program. But I have meet a new problem when I run this program:The diagonal element 1 is zero, the inversion could not be completed because the input matrix is singular.
Have you ever meet this error during runing this program? I look forward to your reply. Thank you!  : )
(I apologize for my broken English.）

hnglp, 2024-04-02:
I haven't had a chance to run it and test it yet. Actually, if you look at it, I get the following error when running the demo. If you have any ideas about this, I would be very happy if you could help me.

```
Traceback (most recent call last):
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\slam.py", line 15, in <module>
    from gui import gui_utils, slam_gui
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\gui\slam_gui.py", line 19, in <module>
    from gui.gl_render import util, util_gau
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\gui\gl_render\__init__.py", line 1, in <module>
    from . import render_ogl, util, util_gau
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\gui\gl_render\render_ogl.py", line 31, in <module>
    raise ImportError
ImportError
```

foreverlong, 2024-04-03:
> from gui import gui_utils, slam_gui

from gui import gui_utils, slam_gui 
maybe you should pip install these models?

ctl6998, 2024-07-11:
> OS: Win 11
>
> I have updated the versions in the dependencies as follows:
>
> ```
> dependencies:
>   - cudatoolkit=11.7
>   - plyfile=0.8.1
>   - python=3.10.14
>   - pip=22.3.1
>   - pytorch=1.13.1
>   - [...]

This method works well for me as I'm using CUDA 11.7 too, thank you!

closed 2024-08-29 by hnglp