muskie82/MonoGS #28: Camera Tracking result on Replica for Monocular SLAM Results

8pterosaur, 2024-03-19:
Hello, the paper shows the results of the RGB-D SLAM on the Replica datasets. 
However, I obtained Monocular SLAM tracking results by adjusting the configuration file. 
For example, on room0 it's about **4cm**, I don't know where I went wrong and the result is unexpected.

Therefore, can you provide the complete evaluation results or guide us on how to adjust configuration file to get correct results.

Thx so much

rmurai0610, 2024-03-19:
Hi, 
The Replica dataset contains many rotational motions, which is challenging for monocular SLAM.
We didn't have much time, and monocular Replica is out of scope for our work as is (and not included in the paper). 
That said, if you are interested, you can probably tweak the hyperparameters to get better performance - maybe with more mapping iterations in a single process mode or adjusting the keyframing parameter/strategy.

8pterosaur, 2024-03-20:
thx, I will try

closed 2024-03-20 by 8pterosaur

suvzzc, 2024-09-19:
> Hello, the paper shows the results of the RGB-D SLAM on the Replica datasets. However, I obtained Monocular SLAM tracking results by adjusting the configuration file. For example, on room0 it's [...]

> Hello, the paper shows the results of the RGB-D SLAM on the Replica datasets. However, I obtained Monocular SLAM tracking results by adjusting the configuration file. For example, on room0 it's [...]

> Hello, the paper shows the results of the RGB-D SLAM on the Replica datasets. However, I obtained Monocular SLAM tracking results by adjusting the configuration file. For example, on room0 it's [...]
Hi, can you tell me what configuration parameters you have adjusted?

suvzzc, 2024-09-19:
> Hello, the paper shows the results of the RGB-D SLAM on the Replica datasets. However, I obtained Monocular SLAM tracking results by adjusting the configuration file. For example, on room0 it's [...]
Hi, can you tell me what configuration parameters you have adjusted?