muskie82/MonoGS #29: Optimization parameters about exposure_a and exposure_b

8pterosaur, 2024-03-19:
While optimizing the camera pose, I noticed that you also optimized `exposure_a` and `exposure_b`, which mainly appear in the following places:
```
image_ab = (torch.exp(viewpoint.exposure_a)) * image + viewpoint.exposure_b
```
Sorry for my lack of knowledge, I have never seen this done before.
Q1: Why did you consider doing this? Can you give me some reference documents?
Q2: Regarding whether the results have been greatly improved, have you done ablation experiments on the Replilca and TMU datasets?

Thx

rmurai0610, 2024-03-19:
Hi, it's a commonly used technique for direct methods (for example, used in DSO)
https://arxiv.org/abs/1607.02565
These affine brightness parameters are optimised to handle varying exposure.
More recently, it's been used in Depth Covariance and Point SLAM
https://arxiv.org/abs/2303.12157
https://arxiv.org/abs/2304.04278

It's not really our contribution, so we didn't perform ablation, but I would expect it to not affect Replica (since it's synthetic) but to affect TUM.

8pterosaur, 2024-03-20:
ok, thx

closed 2024-03-20 by 8pterosaur