muskie82/MonoGS #30: How to visialize pointcloud using GUI after training

Viki1314, 2024-03-20:
Hello! Thanks for your great work.

I wonder how to use GUI to visialize using the GUI you design.
I have the result-----point_cloud.ply file of the fr3_desk scene.

Trank you very much.

muskie82, 2024-03-20:
Hi, thanks for your interest in our work.

So far, GUI is just for online reconstruction.
To visualise the result, please use some other viewer tools following the issue #3

We plan to release the GUI for offline visualisation later. Indeed we have the GUI for offline inspection during development, but it requires some refactoring to be compatible with this open-source version of the codebase.

Thanks,

Viki1314, 2024-03-20:
yes,I saw this issue. Thank for your reply!

closed 2024-03-21 by rmurai0610