muskie82/MonoGS #33: pip install submodules/simple-knn wrong

foreverlong, 2024-03-21:
Thanks for the great work. When I try to create the project env using Win10,I found this wrong :OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
```
(MonoGS) PS D:\MonoGS> pip install submodules/simple-knn
Looking in indexes: https://mirrors.aliyun.com/pypi/simple/
Processing d:\monogs\submodules\simple-knn
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      Traceback (most recent call last):
        File "<string>", line 36, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "D:\MonoGS\submodules\simple-knn\setup.py", line 30, in <module>
          extra_compile_args={"nvcc": [], "cxx": cxx_compiler_flags})
        File "D:\Conda\envs\MonoGS\lib\site-packages\torch\utils\cpp_extension.py", line 983, in CUDAExtension
          library_dirs += library_paths(cuda=True)
        File "D:\Conda\envs\MonoGS\lib\site-packages\torch\utils\cpp_extension.py", line 1105, in library_paths
          paths.append(_join_cuda_home(lib_dir))
        File "D:\Conda\envs\MonoGS\lib\site-packages\torch\utils\cpp_extension.py", line 2125, in _join_cuda_home
          raise EnvironmentError('CUDA_HOME environment variable is not set. '
      OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

muskie82, 2024-03-21:
Thank you for your interest in our work.

Althugh we didn't test the method on Windows, similar issues are reported in original 3D Gaussian Splatting repo.
Can you check and try the solutions there?

https://github.com/graphdeco-inria/gaussian-splatting/issues/520

foreverlong, 2024-03-22:
> Thank you for your interest in our work.
> 
> Althugh we didn't test the method on Windows, similar issues are reported in original 3D Gaussian Splatting repo. Can you check and try the solutions there?
> 
> * [Pip subprocess error at local setup step graphdeco-inria/gaussian-splatting#520](https://github.com/graphdeco-inria/gaussian-splatting/issues/520)

Thank you for your answer! I check it, and this problem is finished. But I meet a new problem. I'm working on it. Thank you.

closed 2024-03-22 by muskie82