muskie82/MonoGS #34: ValueError: Unable to determine Axes to steal space for Colorbar.

z157181773, 2024-03-21:
MonoGS: Evaluating ATE at frame:  106
Eval: RMSE ATE [m] 0.007080079125603161
Traceback (most recent call last):
  File "/home/zzq/workspace/MonoGS/slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "/home/zzq/workspace/MonoGS/slam.py", line 110, in __init__
    self.frontend.run()
  File "/home/zzq/workspace/MonoGS/utils/slam_frontend.py", line 468, in run
    eval_ate(
  File "/home/zzq/workspace/MonoGS/utils/eval_utils.py", line 105, in eval_ate
    ate = evaluate_evo(
  File "/home/zzq/workspace/MonoGS/utils/eval_utils.py", line 54, in evaluate_evo
    evo.tools.plot.traj_colormap(
  File "/home/zzq/.local/lib/python3.10/site-packages/evo/tools/plot.py", line 395, in traj_colormap
    cbar = fig.colorbar(
  File "/home/zzq/.local/lib/python3.10/site-packages/matplotlib/figure.py", line 1285, in colorbar
    raise ValueError(
ValueError: Unable to determine Axes to steal space for Colorbar. Either provide the *cax* argument to use as the Axes for the Colorbar, provide the *ax* argument to steal space from it, or add *mappable* to an Axes.
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Resetting the opacity of non-visible Gaussians

I don't know why this situation occurred. I have installed and run the code according to the requirements. After running, only a few cameras can be seen on the graphical interface, and then this situation will occur. That is to say, an error occurred after running for a while.

z157181773, 2024-03-21:
[image]

muskie82, 2024-03-21:
Hi,

Similar to #18, can you try downgrading matplotlib?
pip install matplotlib==3.5.3

z157181773, 2024-03-21:
zzq@zzq:~/workspace/MonoGS$ python3 slam.py --config configs/rgbd/tum/fr3_office.yaml
MonoGS: saving results in results/datasets_tum/2024-03-21-17-04-46
WARNING: Using soft CircularBuffer (6144 KiB)
FEngine (64 bits) created at 0x7cfe5403c010 (threading is enabled)
FEngine resolved backend: OpenGL
MonoGS: Resetting the system
MonoGS: Initialized map
Process Process-4:
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/zzq/workspace/MonoGS/gui/slam_gui.py", line 688, in run
    app.run()
  File "/home/zzq/workspace/MonoGS/gui/slam_gui.py", line 676, in update
    self.scene_update()
  File "/home/zzq/workspace/MonoGS/gui/slam_gui.py", line 663, in scene_update
    self.render_gui()
  File "/home/zzq/workspace/MonoGS/gui/slam_gui.py", line 655, in render_gui
    results = self.rasterise(current_cam)
  File "/home/zzq/workspace/MonoGS/gui/slam_gui.py", line 564, in rasterise
    rendering_data = render(
  File "/home/zzq/workspace/MonoGS/gaussian_splatting/gaussian_renderer/__init__.py", line 130, in render
    rendered_image, radii, depth, opacity, n_touched = rasterizer(
  File "/home/zzq/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/zzq/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/zzq/.local/lib/python3.10/site-packages/diff_gaussian_rasterization/__init__.py", line 232, in forward
    return rasterize_gaussians(
  File "/home/zzq/.local/lib/python3.10/site-packages/diff_gaussian_rasterization/__init__.py", line 34, in rasterize_gaussians
    return _RasterizeGaussians.apply(
  File "/home/zzq/.local/lib/python3.10/site-packages/torch/autograd/function.py", line 553, in apply
    return super().apply(*args, **kwargs)  # type: ignore[misc]
  File "/home/zzq/.local/lib/python3.10/site-packages/diff_gaussian_rasterization/__init__.py", line 99, in forward
    num_rendered, color, radii, geomBuffer, binningBuffer, imgBuffer, depth, opacity, n_touched = _C.rasterize_gaussians(*args)
torch.cuda.OutOfMemoryError: CUDA out of memory. Tried to allocate 16.00 MiB. GPU 0 has a total capacity of 7.75 GiB of which 17.31 MiB is free. Process 8355 has 1.30 GiB memory in use. Process 11183 has 896.00 MiB memory in use. Process 12110 has 1.26 GiB memory in use. Process 15561 has 3.59 GiB memory in use. Process 16051 has 244.00 MiB memory in use. Including non-PyTorch memory, this process has 204.00 MiB memory in use. Process 16245 has 216.00 MiB memory in use. Of the allocated memory 45.51 MiB is allocated by PyTorch, and 12.49 MiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)
Traceback (most recent call last):
  File "/home/zzq/workspace/MonoGS/slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "/home/zzq/workspace/MonoGS/slam.py", line 110, in __init__
    self.frontend.run()
  File "/home/zzq/workspace/MonoGS/utils/slam_frontend.py", line 392, in run
    render_pkg = self.tracking(cur_frame_idx, viewpoint)
  File "/home/zzq/workspace/MonoGS/utils/slam_frontend.py", line 173, in tracking
    loss_tracking = get_loss_tracking(
  File "/home/zzq/workspace/MonoGS/utils/slam_utils.py", line 60, in get_loss_tracking
    return get_loss_tracking_rgbd(config, image_ab, depth, opacity, viewpoint)
  File "/home/zzq/workspace/MonoGS/utils/slam_utils.py", line 85, in get_loss_tracking_rgbd
    l1_rgb = get_loss_tracking_rgb(config, image, depth, opacity, viewpoint)
  File "/home/zzq/workspace/MonoGS/utils/slam_utils.py", line 70, in get_loss_tracking_rgb
    l1 = opacity * torch.abs(image * rgb_pixel_mask - gt_image * rgb_pixel_mask)
torch.cuda.OutOfMemoryError: CUDA out of memory. Tried to allocate 20.00 MiB. GPU 0 has a total capacity of 7.75 GiB of which 19.31 MiB is free. Process 8355 has 1.30 GiB memory in use. Process 11183 has 896.00 MiB memory in use. Process 12110 has 1.26 GiB memory in use. Process 15561 has 3.59 GiB memory in use. Including non-PyTorch memory, this process has 242.00 MiB memory in use. Process 16131 has 204.00 MiB memory in use. Process 16245 has 216.00 MiB memory in use. Of the allocated memory 80.98 MiB is allocated by PyTorch, and 5.02 MiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)

Thank you for your recovery! But now there is a new problem, the display interface has turned white and then stuck

z157181773, 2024-03-21:
Thank you for your reply! I'll be fine after restarting the computer!

z157181773, 2024-03-21:
Thank you for your reply! It's ready to run!

closed 2024-03-21 by muskie82

JzHuai0108, 2024-09-01:
I bumped up the evo version to 1.19.1, and then change the two line in eval_utils.py from 
```
    traj_est_aligned = trajectory.align_trajectory(
        traj_est, traj_ref, correct_scale=monocular
    )
```
to
```
    traj_est_aligned = copy.deepcopy(traj_est)
    traj_est_aligned.align(traj_ref, correct_scale=monocular)
```
One good side effect is that I do not need to downgrade my matplotlib.