muskie82/MonoGS #35: building "diff-gaussian-rasterization" error: expected an expression detected during instantiation of

foreverlong, 2024-03-22:
Thanks for the great work. When I try to create the project env using Win10,I found this wrong :
D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\auxiliary.h(151): warning #177-D: variable "p_proj" was declared but never referenced

D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(471): warning #177-D: variable "c" was declared but never referenced
          detected during instantiation of "void preprocessCUDA<C>(int, int, int, const float3 *, const int *, const float *, const __nv_bool *, const glm::vec3 *, const glm::vec4 *, float, const float *, const float *, const float *, const glm::vec3 *, const float3 *, glm::vec3 *, float *, float *, float *, float *, glm::vec3 *, glm::vec4 *, float *) [with C=3]"
(838): here

D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(472): warning #177-D: variable "d" was declared but never referenced
          detected during instantiation of "void preprocessCUDA<C>(int, int, int, const float3 *, const int *, const float *, const __nv_bool *, const glm::vec3 *, const glm::vec4 *, float, const float *, const float *, const float *, const glm::vec3 *, const float3 *, glm::vec3 *, float *, float *, float *, float *, glm::vec3 *, glm::vec4 *, float *) [with C=3]"
(838): here

D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(554): error: expected an expression
          detected during instantiation of "void renderCUDA<C>(const uint2 *, const uint32_t *, int, int, const float *, const float2 *, const float4 *, const float *, const float *, const float *, const uint32_t *, const float *, const float *, float3 *, float4 *, float *, float *, float *) [with C=3U]"
(882): here

yanyan-li, 2024-03-22:
We have the same issue using Win10

> Thanks for the great work. When I try to create the project env using Win10,I found this wrong : D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\auxiliary.h(151): warning #177-D: variable "p_proj" was declared but never referenced
> 
> D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(471): warning #177-D: variable "c" was declared but never referenced detected during instantiation of "void preprocessCUDA(int, int, int, const float3 *, const int *, const float *, const __nv_bool *, const glm::vec3 *, const glm::vec4 *, float, const float *, const float *, const float *, const glm::vec3 *, const float3 *, glm::vec3 *, float *, float *, float *, float *, glm::vec3 *, glm::vec4 *, float *) [with C=3]" (838): here
> 
> D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(472): warning #177-D: variable "d" was declared but never referenced detected during instantiation of "void preprocessCUDA(int, int, int, const float3 *, const int *, const float *, const __nv_bool *, const glm::vec3 *, const glm::vec4 *, float, const float *, const float *, const float *, const glm::vec3 *, const float3 *, glm::vec3 *, float *, float *, float *, float *, glm::vec3 *, glm::vec4 *, float *) [with C=3]" (838): here
> 
> D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(554): error: expected an expression detected during instantiation of "void renderCUDA(const uint2 *, const uint32_t *, int, int, const float *, const float2 *, const float4 *, const float *, const float *, const float *, const uint32_t *, const float *, const float *, float3 *, float4 *, float *, float *, float *) [with C=3U]" (882): here

The issue was fixed here:
https://github.com/muskie82/MonoGS/issues/16#issuecomment-1989798377

foreverlong, 2024-03-27:
> We have the same issue using Win10
> 
> > Thanks for the great work. When I try to create the project env using Win10,I found this wrong : D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\auxiliary.h(151): warning #177-D: variable "p_proj" was declared but never referenced
> > D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(471): warning #177-D: variable "c" was declared but never referenced detected during instantiation of "void preprocessCUDA(int, int, int, const float3 *, const int *, const float *, const __nv_bool *, const glm::vec3 *, const glm::vec4 *, float, const float *, const float *, const float *, const glm::vec3 *, const float3 *, glm::vec3 *, float *, float *, float *, float *, glm::vec3 *, glm::vec4 *, float *) [with C=3]" (838): here
> > D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(472): warning #177-D: variable "d" was declared but never referenced detected during instantiation of "void preprocessCUDA(int, int, int, const float3 *, const int *, const float *, const __nv_bool *, const glm::vec3 *, const glm::vec4 *, float, const float *, const float *, const float *, const glm::vec3 *, const float3 *, glm::vec3 *, float *, float *, float *, float *, glm::vec3 *, glm::vec4 *, float *) [with C=3]" (838): here
> > D:\MonoGS\submodules\diff-gaussian-rasterization\cuda_rasterizer\backward.cu(554): error: expected an expression detected during instantiation of "void renderCUDA(const uint2 *, const uint32_t *, int, int, const float *, const float2 *, const float4 *, const float *, const float *, const float *, const uint32_t *, const float *, const float *, float3 *, float4 *, float *, float *, float *) [with C=3U]" (882): here
> 
> The issue was fixed here: [#16 (comment)](https://github.com/muskie82/MonoGS/issues/16#issuecomment-1989798377)

Thank you! That works!

closed 2024-03-29 by muskie82