muskie82/MonoGS #36: The visualization of Gaussians is strange

yanyan-li, 2024-03-22:
Thank you for the great work.
When I run the system, the Gaussian plot looks very strange.
Does anyone know the cause of this problem.
THX.

[image]

muskie82, 2024-03-22:
Hi, thanks for trying our code.

Hmm, I've never seen this before.
- This is the default color rendering, right? 
- Does the issue happen only in rgbd/fr3 or in all configs?

yanyan-li, 2024-03-22:
Hi. Thanks for your reply. 
I tried _f3_strtex_far_ and _f1_desk_. The similar phenomenon is witnessed in these sequences.
And some issues in my trying in Win10, like 
`File "E:\MonoGS-main\utils\slam_utils.py", line 120, in get_loss_mapping_rgbd
    dtype=torch.float32, device=image.device
RuntimeError: CUDA error: an illegal memory access was encountered
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing `CUDA_LAUNCH_BLOCKING=1.` 
`File "E:\MonoGS-main\gaussian_splatting\utils\graphics_utils.py", line 41, in getWorld2View2
    C2W = torch.linalg.inv(Rt)
torch._C._LinAlgError: torch.linalg.inv: The diagonal element 1 is zero, the inversion could not be completed because the input matrix is singular.
[W C:\cb\pytorch_1000000000000\work\torch\csrc\CudaIPCTypes.cpp:15] Producer process has been terminated before all shared CUDA tensors released. See Note [Sharing CUDA tensors]`

Maybe some problems in my environment. I will try to find out the reason.

muskie82, 2024-03-22:
Thanks.
I think the error message looks similar to #20, can you check it?

SClerckx, 2024-03-30:
[image]
Same issue here.
Sometimes I do get splats but with unstable behaviour:
[image]

This is the console output:
`MonoGS: saving results in results\datasets_tum\2024-03-30-22-27-03
FEngine (64 bits) created at 000001FEB6128050 (threading is enabled)
FEngine resolved backend: OpenGL
MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Keyframes lacks sufficient overlap to initialize the map, resetting.
MonoGS: Resetting the system
MonoGS: Initialized map`

I have also seen the torch.linalg.inv error but it doesn't occur in every run.
`torch.linalg.inv: The diagonal element 1 is zero, the inversion could not be completed because the input matrix is singular.`

My environment: Windows11, CUDA 11.7, Pytorch 1.12

I'm currently trying to find a lower pytorch and cuda version combo that hopefully doesn't have this problem

foreverlong, 2024-04-01:
> ![图像](https://private-user-images.githubusercontent.com/36774906/318225105-0fd9ff04-98d0-4f6f-a122-884f512a325c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5MzQ2OTIsIm5iZiI6MTcxMTkzNDM5MiwicGF0aCI6Ii8zNjc3NDkwNi8zMTgyMjUxMDUtMGZkOWZmMDQtOThkMC00ZjZmLWExMjItODg0ZjUxMmEzMjVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAxVDAxMTk1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFkODU3MzI2NjdiYTY1Mjg4NjFlNzk2ZmIxYTgxZjU4NTQ1M2Y5N2M4Njk5MGQzMWJiZTg1YWU0YTg1ZTYxZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RTycftbyQ9xwrG2t_zRq6d9PeoQcLnL-AZmhuCAWHwM) 这里同样的问题。 有时我确实会遇到一些问题，但不稳定： ![图像](https://private-user-images.githubusercontent.com/36774906/318225434-f9d57d77-824e-49a8-b15b-b64b8d9f07d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5MzQ2OTIsIm5iZiI6MTcxMTkzNDM5MiwicGF0aCI6Ii8zNjc3NDkwNi8zMTgyMjU0MzQtZjlkNTdkNzctODI0ZS00OWE4LWIxNWItYjY0YjhkOWYwN2QzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAxVDAxMTk1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiNGMyZDAxNjMwNDcwYWMwYmU2ODc2ZmEwNThiNjg1YzYxZDEwZjE2ODM5NzQ1OTVkZmZjN2RlYWVkNjVjMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XEuETkgTqqlJvPewAkmW7zGvTMZVRieKIU7P2jrtU0k)
> 
> 这是控制台输出： `MonoGS: saving results in results\datasets_tum\2024-03-30-22-27-03 FEngine (64 bits) created at 000001FEB6128050 (threading is enabled) FEngine resolved backend: OpenGL MonoGS: Resetting the system MonoGS: Initialized map MonoGS: Resetting the opacity of non-visible Gaussians MonoGS: Keyframes lacks sufficient overlap to initialize the map, resetting. MonoGS: Resetting the system MonoGS: Initialized map`
> 
> 我还看到了 torch.linalg.inv 错误，但它不会在每次运行中发生。 `torch.linalg.inv: The diagonal element 1 is zero, the inversion could not be completed because the input matrix is singular.`
> 
> 我的环境：Windows11、CUDA 11.7、Pytorch 1.12
> 
> 我目前正在尝试找到一个较低的 pytorch 和 cuda 版本组合，希望不会出现这个问题

I have this issue too. Have you ever fix it?

dumyCq, 2024-04-22:
I am using Windows 10 and I have the same issue, I am using Pytorch 1.9.0 and CUDA 11.1. If someone wants to collaborate on this, feel free to contact me.

dumyCq, 2024-04-22:
> I have this issue too. Have you ever fix it?

I am using Windows 10 and I have the same issue, I am using Pytorch 1.9.0 and CUDA 11.1. If someone wants to collaborate on this, feel free to contact me. I have wechat DarrenPSU