muskie82/MonoGS #37: Failed to install diff_gaussian_rasterization on Windows (Solved)

wangwang-xyz, 2024-03-23:
# system environment
* Windows 10
* Visual Studio 2022
* cuda==11.8.0
* pytorch==2.0.0

# Error message
I get the following error message when I run `pip install. \submodules\diff-gaussian-rasterization-w-pose\`

```shell
D:\GaussianSLAM\MonoGS\submodules\diff-gaussian-rasterization-w-pose\cuda_rasterizer\backward.cu(554): error: expected an expression
                detected during instantiation of "void renderCUDA<C>(const uint2 *, const uint32_t *, int, int, const float *, const float2 *, 
const float4 *, const float *, const float *, const float *, const uint32_t *, const float *, const float *, float3 *, float4 *, float *, float *, float *) 
[with C=3U]"
      (882): here

      1 error detected in the compilation of "D:/GaussianSLAM/MonoGS/submodules/diff-gaussian-rasterization-w-pose/cuda_rasterizer/backward.cu".

      backward.cu
```
In line `554` of `/cuda_rasterizer/backward.cu`

```
548    template <typename group_t, typename... Lists>
549    __device__ void render_cuda_reduce_sum(group_t g, Lists... lists) {
550      int lane = g.thread_rank();
551      g.sync();
552    
553      for (int i = g.size() / 2; i > 0; i /= 2) {
554        (...,
555         reduce_helper(                                      //  ERROR HERE
556             lane, i, lists)); // Fold expression: apply reduce_helper for each list
557        g.sync();
558      }
559   }
```

# Solution
Modify the `render_cuda_reduce_sum` function as follows

```
template <typename group_t, typename... Lists>
__device__ void render_cuda_reduce_sum(group_t g, Lists... lists) {
  int lane = g.thread_rank();
  g.sync();

  for (int i = g.size() / 2; i > 0; i /= 2) {
//     (...,reduce_helper(lane, i, lists)); // Fold expression: apply reduce_helper for each list
    (void)std::initializer_list<int>{(reduce_helper(lane, i, lists), 0)...};
    g.sync();
  }
}
```

muskie82, 2024-03-23:
Thanks for sharing your solution!

closed 2024-03-23 by muskie82

dumyCq, 2024-04-22:
> # system environment
> * Windows 10
> * Visual Studio 2022
> * cuda==11.8.0
> * pytorch==2.0.0
>
> # Error message
> I get the following error message when I run `pip install. [...]

really appreciate! did you successfully have the system run in windows? shall I ask you some questions related to it's strange performance

wangwang-xyz, 2024-04-23:
> really appreciate! did you successfully have the system run in windows? shall I ask you some questions related to it's strange performance

I didn't successfully run the original `MonoGS` (multiprocessing version) on `windows` or `wsl2`. The main reason is that I encountered the following two bugs that I could not solve

1. `torch.multiprocessing` weirdly sets shared variables to 0 (like camera pose).
2. `img.cuda()` is stuck while loading the second image (possibly due to a thread deadlock).

So, I changed `MonoGS` to a single-process version and simply coded a GUI for it. The modified `MonoGS` was able to run on `wsl2`. (`windows` was not tested.)

But I highly recommend running `MonoGS` on `Ubuntu`. You won't run into any of the weird bugs above, and the experience will be much better.