muskie82/MonoGS #38: ImportError: Cannot load backend 'TkAgg' which requires the 'tk' interactive framework, as 'headless' is currently running

kk6398, 2024-03-24:
A error occurs when  “ python slam.py --config configs/mono/tum/fr3_office.yaml” 

[image]

kk6398, 2024-03-24:
I use SSH to remotely connect to the Linux terminal. The vscode shell I use does not have a graphical interface. Is this the reason?

kk6398, 2024-03-24:
Yep, I try run "python slam.py --config configs/mono/tum/fr3_office.yaml" on the unbuntu graphical interface, and the error "Cannot load backend 'TkAgg' which requires the 'tk' interactive framework, as 'headless' is currently running" doesnot appear.  

However, anthor question occurs: ①there is no gui window pop out, and ②"cuda out of memory" when processing"Evaluating ATE at frame:  2230".     (I run the code on 4090 24G)    

What is the reason and how I deal with it?

muskie82, 2024-03-25:
Hi,

Never seen this before, but I guess the GUI is dead for some reason and the q_main2vis is not cleaned up, which leads to out of memory.
Do you see any suspicious error message related to GUI?

kk6398, 2024-03-25:
okay, I display the code log as follows:
`python slam.py --config configs/mono/tum/fr1_desk.yaml 
MonoGS: saving results in 
results/tum_rgbd_dataset_freiburg1_desk/2024-03-24-21-07-31
FEngine (64 bits) created at 0x53785400 (threading is enabled)
FEngine resolved backend: OpenGL
X Error of failed request:  GLXBadFBConfig
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  34 ()
  Serial number of failed request:  39
  Current serial number in output stream:  37
MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Keyframes lacks sufficient overlap to initialize the map, resetting.
MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Performing initial BA for initialization
MonoGS: Initialized SLAM
MonoGS: Evaluating ATE at frame:  74
Eval: RMSE ATE [m] 0.008595195845623836
MonoGS: Evaluating ATE at frame:  123
Eval: RMSE ATE [m] 0.01270222476891415
MonoGS: Evaluating ATE at frame:  164
Eval: RMSE ATE [m] 0.01736790069366283
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Evaluating ATE at frame:  203
Eval: RMSE ATE [m] 0.022410050076178276
MonoGS: Evaluating ATE at frame:  257
Eval: RMSE ATE [m] 0.029316772769155855
MonoGS: Evaluating ATE at frame:  294
Eval: RMSE ATE [m] 0.033102492927375486
MonoGS: Evaluating ATE at frame:  323
Eval: RMSE ATE [m] 0.03582281334425646
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Evaluating ATE at frame:  367
Eval: RMSE ATE [m] 0.03840116612275182
MonoGS: Evaluating ATE at frame:  399
Eval: RMSE ATE [m] 0.0379590706327925
MonoGS: Evaluating ATE at frame:  451
Eval: RMSE ATE [m] 0.03892031242012154
MonoGS: Evaluating ATE at frame:  494
Eval: RMSE ATE [m] 0.03823152505622439
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Evaluating ATE at frame:  525
Eval: RMSE ATE [m] 0.038492287772041056
Eval: RMSE ATE [m] 0.03811678444826245
Eval: Total time 462.99578125
Eval: Total FPS 1.2786293611611608
MonoGS: Backend stopped and joined the main thread
MonoGS: GUI Stopped and joined the main thread
MonoGS: Done.
`

kk6398, 2024-03-25:
okay, I display the code log as follows:
python slam.py --config configs/mono/tum/fr1_desk.yaml
[image]
[image]

muskie82, 2024-03-25:
I think `x error of failed request` is the one related.
Your environment seeems to have some issue with OpenGL, which prevents GUI launch.

Can you try to fix the error message?
As a quick search, this might be relevant.
https://askubuntu.com/questions/1369900/x-error-of-failed-request-glxbadfbconfig

kk6398, 2024-03-25:
Thank for your reply. I tried entering "export MESA_GL_VERSION_OVERRIDE=4.3" or export "MESA_GL_VERSION_OVERRIDE=4.5" as the ISSUE#5, but the window MonoGS only flashed for one second and suddenly exiting, then reporting an error as follow. I hope you can answer.   The version of pyopengl is 3.1.7   cry~ ~    
[image]

kk6398, 2024-03-25:
> Thank for your reply. I tried entering "export MESA_GL_VERSION_OVERRIDE=4.3" or export "MESA_GL_VERSION_OVERRIDE=4.5" as the ISSUE#5, but the window MonoGS only flashed for one second and suddenly [...]

I may not have fully stated the environment:   ubuntu 18.04   GPU: 4090   cuda11.6  pytorch1.12.1   pyopengl 3.1.7

muskie82, 2024-03-25:
I don't have an immediate answer unfortunately. The error message `x error of failed request` has many reports online, so I recommend searching the web for similar cases and trying out the suggested solutions.

For a temporary solution, you could comment out all the code related to `gl_render` in `slam_gui.py`. `gl_render` is used just for the Ellipsoid shader, so the rest of the GUI should function without it.

kk6398, 2024-03-26:
Hi, I try lots of methods, and successfully run the MonoGS GUI window through [https://github.com/isl-org/Open3D/issues/4429], but some errors occur: there is no picture appears in the left frame. Moreover, there is no response when l click any button on the right.
[image]

muskie82, 2024-03-26:
Any error message?

Also, I recommend you to try some of the Open3D GUI examples, for example [this](https://github.com/isl-org/Open3D/blob/main/examples/python/visualization/demo_scene.py). If the same problem happens, then probably some similar issues/solutions are reported on Open3D repo.

yaochengrong, 2024-03-28:
same issue here

kk6398, 2024-03-28:
> same issue here

I have not solved it yet. Do you have any good methods?

peterWon, 2024-03-29:
@kk6398 I met this error. I solved this problem by 'evo_config set plot_backend agg'

kk6398, 2024-03-29:
> @kk6398 I met this error. I solved this problem by 'evo_config set plot_backend agg'

Can you explain the operation steps in detail please?

whwh747, 2024-04-03:
i meet the same error in vscode ， and the i run this code in mobaxterm ， it is normal @kk6398

kk6398, 2024-04-03:
> i meet the same error in vscode ， and the i run this code in mobaxterm ， it is normal @kk6398

Thank for your reply. I have run this code in mobaxterm.  However, the real-time performance of SLAM displayed in the pop-up window is poor, and it does not perform tracking and mapping as quickly as the video provided by the author.

whwh747, 2024-04-03:
i agree , but i did not find the reason, and my code seems to get stuck after running
[image]
@kk6398

rmurai0610, 2024-04-03:
To solve 
> Cannot load backend 'TkAgg' which requires the 'tk' interactive framework, as 'headless' is currently running

As @peterWon mentions, if you run the command in the terminal,
`evo_config set plot_backend agg`
agg will be set for the backend for the evo. (evo has a global config file, and it's probably set to TkAgg for some reason)

@kk6398 @whwh747, which sequences are you running? 
@whwh747 I cannot replicate the 100% CPU utilisation. What CPU are you using?

whwh747, 2024-04-04:
rgbd model fr3 long office ， and you can clearly see the CPU usage and which processes are using the CPU in nvitop. @rmurai0610 
[image]