muskie82/MonoGS #40: How the camera pose is estimated?

Yoona12, 2024-03-26:
Thanks for your great work ! 
When I read the papre, I find that there is a process of camera pose estimation in SLAM system overview. So  I'm trying to know how the camera pose is estimated. However , I just find how the camera pose is optimised but not estimated when reading the papre.
So I'm confused that how the camera pose is estimated ?
This may be a stupid question, but I would like to get an explanation from you.
Thanks !

muskie82, 2024-03-26:
Hi, thanks for your interest.
"estimation" and "optimisation" is basically the same, we get the current camera pose estimation by optimising the initial camera pose (the estimated camera pose of the previous frame) iteratively.

Yoona12, 2024-03-27:
Thanks for your reply!