muskie82/MonoGS #41: Questions about code writing

z157181773, 2024-03-26:
Hello, sorry to bother you, I am looking at the code recently and I have two questions
Q1: Regarding the process of running the code, I found that if I use a monocular photo without depth (that is, RGB), the depth parameter is not used in the slam process. So if I run a Depth estimation estimates the depth without monocular photos (RGB). If this is similar to RGBD for slam, will the effect be better?
Q2: I want to improve my ability to write code, so I want to write code based on your code. I would like to ask if there is a good direction for my reference!
Thank you for your hard work!

muskie82, 2024-03-26:
Hi, thank you for your interest.

1. I don't understand your question well, can you explain a bit more detail? Are you asking if monocular depth estimation from learned network improves our monocular SLAM performance? If monocular depth estimation is accurate, then I think yes.
2. In general, I think it is a matter of experience. Copy and hack someone's code, write your own, reviewed by others... just iterating this process. In our codebase specifically, we use standard tracking/mapping framework which runs with multi-thread/process. [pyslam](https://github.com/luigifreda/pyslam) could be a good starting point to get familiar with SLAM example written in Python.

z157181773, 2024-03-26:
thank you for your reply!
Regarding question 1: I mean, whether learning the network's monocular depth estimation can improve monocular SLAM performance. Maybe there is something wrong with my expression, I'm sorry for confusing you!
Regarding question 2: Thank you very much for your answer. I am currently in the learning stage. I have just started to contact 3dgs. I have studied SLAM before and want to see how the two can be better combined. Your paper has given me a lot of inspiration. Inspire. But since I am currently in the beginner stage and groping for myself, thank you for your advice.
Thank you very much for your efforts so that I can learn more!
(My English skills are not very good, so there may be problems with my expression, I'm sorry)

muskie82, 2024-03-26:
No problem, I understand.
Our monocular SLAM is minimal setup, and the performance could be improved by leveraging monocular predictions (depth, normal, flow, etc..). It is not as straightforward as depth sensor, because monocular network tends to struggle with temporal consistency. The way to leverage these monocular cues in our monocular SLAM is an interesting research direction.

z157181773, 2024-03-27:
Hello! Sorry to bother you again. After reading the code about RGB depth, I found some confusion:
Q1: In the frontend's util, there is a function called 'def add new_keyframe (self, cur_frame_idx, depth=None, opacity=None, init=False)' with an init parameter, but I haven't found any specific logic triggered by the value of this init parameter
Q2: At line 124 of the code, depth_map=self. add new_keyframe (cur_frame_idx, init=True), and the init parameter seems to be of little use.

z157181773, 2024-03-27:
Q3: I often see monocular in the code, does this refer to the input RGB image? (No depth input)
Looking forward to your reply! Thank you!

muskie82, 2024-03-27:
Hi,
1&2 Yeah, it is not used in the fuction at all. We will fix it, thanks! Some variables used during development seems to remain in the current codebase.
3 Yes, it is either RGB or Grayscale image without depth input.  

Thanks

z157181773, 2024-03-31:
Hello author, sorry to bother you again!    I have two questions about the MonoGS and gaussian splatting code:
Q1: In gaussian splatting, the main operation is train.py, but I noticed that there is no train.py file in MonoGS, I know that the process of gaussian splatting is divided into the slam section.    My question is that in MonoGS, Is the code modified according to train.py in the original gaussian splatting and add it to MonoGS?
Q2: In MonoGS, I found that only slam.py and slam_backend.py had the code about gaussian splatting, while slam_frontend.py only had rendering and visualization, without any iteration about gaussians, right?
Looking forward to your reply! Thank you!

muskie82, 2024-03-31:
These are two different codebase, thus it is hard to tell the exact code-level correspondance.
Our code is based on 3DGS, but treat it as a different one.

Since the initial question of this issue is solved, I will close this issue.
It is not common to keep posting different questions in one GitHub issue.

closed 2024-03-31 by muskie82