muskie82/MonoGS #42: extend to object 6dof tracking

nuaayxy, 2024-03-26:
Hi,

Awesome work.
I wonder how feasible it is to extend it to handle object pose estimation using this.
Sounds like we can do an inverted slam for the object if we have masked segmented objects in multiple images?

muskie82, 2024-03-27:
Hi, thanks for your interest in our work.

I think it is doable.
The task sounds similar to [iNeRF](https://yenchenlin.me/inerf/), and finding an unique property of 3DGS for the task is an interesting topic.

nuaayxy, 2024-03-27:
how will it handle occlusion from hand if i use hand and interact/rotate the object and trying to get the pose of object.

nuaayxy, 2024-03-28:
> Hi, thanks for your interest in our work.
> 
> I think it is doable. The task sounds similar to [iNeRF](https://yenchenlin.me/inerf/), and finding an unique property of 3DGS for the task is an interesting topic.

what i mean if we do not have the nerf/3d model first but I just want same method as the monoGS to optimize and get object pose while at same get gaussian splats or nerf

muskie82, 2024-03-28:
Hi,
I am not sure if I understand the task correctly, but something like [BundleSDF](https://bundlesdf.github.io/)?

nuaayxy, 2024-03-28:
yes, something like that but with 3dgs

muskie82, 2024-03-28:
ok, so using SLAM for object tracking and mapping instead of camera ego-motion tracking, right?
Something similar should also be done by 3DGS as NeRF/NeuralSDF can do. Not sure how to deal with hand occlusion, as I am not familiar with the topic. 

BundleSDF paper will be helpful for such technical detail.

Best.

nuaayxy, 2024-03-28:
> ok, so using SLAM for object tracking and mapping instead of camera ego-motion tracking, right? Something similar should also be done by 3DGS as NeRF/NeuralSDF can do. Not sure how to deal with hand [...]

yeah,like object oriented slam using 3dgs. I wonder if someone has done it.

nuaayxy, 2024-03-28:
because most of object scan or tracking still need camera pose, but if it can be solved using your approach it can reconstruct and track the object at the same time that will be good

closed 2024-04-15 by muskie82