muskie82/MonoGS #43: BrokenPipeError: [Errno 32] Broken pipe

Runyu-Zhou05, 2024-03-27:
Although this error does not terminate the program, I'm concerned with this issue.

Platform: `Ubuntu 22.04.4 LTS`
GPU: `RTX 3090` (GPU memory: 24GiB)
Python version: `3.11.8`

My command: `python slam.py --config configs/mono/tum/fr1_desk.yaml --eval`

```python
MonoGS: Running MonoGS in Evaluation Mode
MonoGS: Following config will be overriden
MonoGS:         save_results=True
MonoGS:         use_gui=False
MonoGS:         eval_rendering=True
MonoGS:         use_wandb=True
MonoGS: saving results in results/tum_rgbd_dataset_freiburg1_desk/2024-03-27-16-45-34
wandb: Currently logged in as: xxx (xxxteam). Use `wandb login --relogin` to force relogin
wandb: wandb version 0.16.5 is available!  To upgrade, please run:
wandb:  $ pip install wandb --upgrade
wandb: Tracking run with wandb version 0.16.4
wandb: Run data is saved locally in /media/omnisky/sdc/2021/zry/GS/gs-slam/wandb/run-20240327_164536-5b9yasoj
wandb: Run `wandb offline` to turn off syncing.
wandb: Syncing run configs/mono/tum/fr1_desk_2024-03-27-16-45-34
wandb: ⭐️ View project at https://wandb.ai/xxxteam/MonoGS
wandb: 🚀 View run at https://wandb.ai/xxxteam/MonoGS/runs/5b9yasoj
MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Keyframes lacks sufficient overlap to initialize the map, resetting.
MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Performing initial BA for initialization
MonoGS: Initialized SLAM
MonoGS: Evaluating ATE at frame:  80
Eval: RMSE ATE [m] 0.0075907673407301125
Traceback (most recent call last):
  File "/media/omnisky/sdc/2021/zry/GS/gs-slam/slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/omnisky/sdc/2021/zry/GS/gs-slam/slam.py", line 110, in __init__
    self.frontend.run()
  File "/media/omnisky/sdc/2021/zry/GS/gs-slam/utils/slam_frontend.py", line 468, in run
    eval_ate(
  File "/media/omnisky/sdc/2021/zry/GS/gs-slam/utils/eval_utils.py", line 105, in eval_ate
    ate = evaluate_evo(
          ^^^^^^^^^^^^^
  File "/media/omnisky/sdc/2021/zry/GS/gs-slam/utils/eval_utils.py", line 54, in evaluate_evo
    evo.tools.plot.traj_colormap(
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/evo/tools/plot.py", line 395, in traj_colormap
    cbar = fig.colorbar(
           ^^^^^^^^^^^^^
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/matplotlib/figure.py", line 1285, in colorbar
    raise ValueError(
ValueError: Unable to determine Axes to steal space for Colorbar. Either provide the *cax* argument to use as the Axes for the Colorbar, provide the *ax* argument to steal space from it, or add *mappable* to an Axes.
wandb: 🚀 View run configs/mono/tum/fr1_desk_2024-03-27-16-45-34 at: https://wandb.ai/xxxteam/MonoGS/runs/5b9yasoj
wandb: ️⚡ View job at https://wandb.ai/xxxteam/MonoGS/jobs/QXJ0aWZhY3RDb2xsZWN0aW9uOjE1MzAxMjMyMw==/version_details/v0
wandb: Synced 6 W&B file(s), 0 media file(s), 0 artifact file(s) and 0 other file(s)
wandb: Find logs at: ./wandb/run-20240327_164536-5b9yasoj/logs
Exception in thread IntMsgThr:
Traceback (most recent call last):
Exception in thread NetStatThr:
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
Traceback (most recent call last):
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
    self.run()
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/wandb_run.py", line 300, in check_internal_messages
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/wandb_run.py", line 268, in check_network_status
    self._loop_check_status(
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/wandb_run.py", line 224, in _loop_check_status
    self._loop_check_status(
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/wandb_run.py", line 224, in _loop_check_status
    local_handle = request()
    local_handle = request()
                                      ^^^^^^^^^^^^^^^
^^  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/interface/interface.py", line 795, in deliver_network_status
^
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/interface/interface.py", line 803, in deliver_internal_messages
    return self._deliver_network_status(status)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/interface/interface_shared.py", line 500, in _deliver_network_status
    return self._deliver_internal_messages(internal_message)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/interface/interface_shared.py", line 506, in _deliver_internal_messages
    return self._deliver_record(record)
    return self._deliver_record(record)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/interface/interface_shared.py", line 449, in _deliver_record
    handle = mailbox._deliver_record(record, interface=self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/interface/interface_shared.py", line 449, in _deliver_record
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/mailbox.py", line 455, in _deliver_record
    handle = mailbox._deliver_record(record, interface=self)
    interface._publish(record)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/mailbox.py", line 455, in _deliver_record
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/interface/interface_sock.py", line 51, in _publish
    interface._publish(record)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/interface/interface_sock.py", line 51, in _publish
    self._sock_client.send_record_publish(record)
    self._sock_client.send_record_publish(record)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/sock_client.py", line 221, in send_record_publish
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/sock_client.py", line 221, in send_record_publish
    self.send_server_request(server_req)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/sock_client.py", line 155, in send_server_request
    self.send_server_request(server_req)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/sock_client.py", line 155, in send_server_request
    self._send_message(msg)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/sock_client.py", line 152, in _send_message
    self._send_message(msg)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/sock_client.py", line 152, in _send_message
    self._sendall_with_error_handle(header + data)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/sock_client.py", line 130, in _sendall_with_error_handle
    self._sendall_with_error_handle(header + data)
  File "/media/omnisky/sdc/2021/zry/anaconda3/envs/zrygs/lib/python3.11/site-packages/wandb/sdk/lib/sock_client.py", line 130, in _sendall_with_error_handle
    sent = self._sock.send(data)
           ^^^^^^^^^^^^^^^^^^^^^
BrokenPipeError: [Errno 32] Broken pipe
    sent = self._sock.send(data)
           ^^^^^^^^^^^^^^^^^^^^^
BrokenPipeError: [Errno 32] Broken pipe
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Resetting the opacity of non-visible Gaussians
```

It has not stopped execution until I lost connection to the server after running for 2 hours, while in the paper it is mentioned that rendering is real-time.

Thank you very much.

muskie82, 2024-03-27:
Hi,
can you try downgrading matplotlib?
pip install matplotlib==3.5.3

Runyu-Zhou05, 2024-03-28:
Hi there,

Thanks for your reminder. I've solved the problem by downgrading Python to `3.7.13` as indicated in `environment.yml`, and reinstalling all dependencies. It turned out that a low Python version is actually crucial.

closed 2024-03-28 by Runyu-Zhou05