muskie82/MonoGS #44: Is there any loop closing or Bundle adjustment ?

lbg030, 2024-03-27:
Thank you for your great work!

Since I'm a undergraduated student, my questions may be clumsy. I appreciate it if you can take this into consideration.

The paper said "it didn't use loop closing" and even in evaluation, the models that used loop closing performed much better, so why didn't you use it?

Is the diffrerent purpose of the model ?

muskie82, 2024-03-27:
Hi, thank you for your interest!

Loop closure is definitely nice to have, but it is another big research topic.
It requires deisning algorithms for loop detection, pose drift correction and map update in 3D Gaussian's framework.
So this work focused on SLAM without loop closure, and left it for the future work.

lbg030, 2024-03-27:
Thank you for kind reply.
I have another question.

I know that COLMAP have bundle adjustment function and gaussian splatting(https://github.com/graphdeco-inria/gaussian-splatting) uses COLMAP bundle adjustment.

then, COLMAP functions can't used in GS SLAM?

Thank you for answering my inexperienced question.

muskie82, 2024-03-27:
Hi,

While gaussian splatting uses COLMAP for pose and Gaussian's location initialisation with unordered image inputs, our work is for joint estimation of pose and 3D Gaussians for sequential video image inputs. These are different task.

Hope this clears.

lbg030, 2024-03-27:
thank you very much

this issue is closed

closed 2024-03-27 by lbg030