muskie82/MonoGS #46: Why delta poses have gradients?

icameling, 2024-03-29:
Hello 
Thanks for your excellent job. While delving into the tracking module, I have a question regarding the pose update process. I noticed that the parameters to be updated include the delta pose and exposure parameters. I understand that exposure parameters are involved in calculating the RGB loss. However, I couldn't find where the delta pose is utilized during rendering or loss computation.
After performing backward and step operations, the pose is updated with gradient disabled. Consequently, should there be no gradient concerning the delta pose?

I am really appreciate if you could provide some guidanceon how the delta pose is optimized within the tracking.

Best regards

 

When we backward, the loss computation is independent of the delta pose (cam_rot_delta and cam_trans_delta)
https://github.com/muskie82/MonoGS/blob/main/utils/slam_frontend.py#L162
```
pose_optimizer = torch.optim.Adam(opt_params)
for tracking_itr in range(self.tracking_itr_num):
    render_pkg = render(
        viewpoint, self.gaussians, self.pipeline_params, self.background
    )
    image, depth, opacity = (
        render_pkg["render"],
        render_pkg["depth"],
        render_pkg["opacity"],
    )
    pose_optimizer.zero_grad()
    loss_tracking = get_loss_tracking(
        self.config, image, depth, opacity, viewpoint
    )
    # note
    loss_tracking.backward()

    with torch.no_grad():
        pose_optimizer.step()
        converged = update_pose(viewpoint)
```

`theta` and `rho` are left unused in the rendering.
https://github.com/rmurai0610/diff-gaussian-rasterization-w-pose/blob/main/diff_gaussian_rasterization/__init__.py#L48C1-L87C10
```
class _RasterizeGaussians(torch.autograd.Function):
    @staticmethod
    def forward(
        ctx,
        means3D,
        means2D,
        sh,
        colors_precomp,
        opacities,
        scales,
        rotations,
        cov3Ds_precomp,
        theta,
        rho,
        raster_settings,
    ):

        # Restructure arguments the way that the C++ lib expects them
        args = (
            raster_settings.bg,
            means3D,
            colors_precomp,
            opacities,
            scales,
            rotations,
            raster_settings.scale_modifier,
            cov3Ds_precomp,
            raster_settings.viewmatrix,
            raster_settings.projmatrix,
            raster_settings.projmatrix_raw,
            raster_settings.tanfovx,
            raster_settings.tanfovy,
            raster_settings.image_height,
            raster_settings.image_width,
            sh,
            raster_settings.sh_degree,
            raster_settings.campos,
            raster_settings.prefiltered,
            raster_settings.debug,
        )
```

rmurai0610, 2024-03-29:
Hi thank you for your interest

We compute the gradient in CUDA, using analytical Jacobian rather than autodiff for efficiency (similar motivation as original 3DGS)
https://github.com/rmurai0610/diff-gaussian-rasterization-w-pose

rho and theta are left unused, but we compute the gradient in the actual cuda code and is returned in the backward pass.
Hope this helps!

icameling, 2024-03-30:
Thanks for your quick reply. That solves my question.

closed 2024-03-30 by icameling