muskie82/MonoGS #48: An error occurred when i run the code

marcoastro, 2024-04-01:
When I run the command: python slam.py --config configs/mono/tum/fr3_office.yaml
An error occurred:
MonoGS: saving results in results/datasets_tum/2024-04-01-19-33-17
WARNING: Using soft CircularBuffer (6144 KiB)
FEngine (64 bits) created at 0x55a3be464590 (threading is enabled)
FEngine resolved backend: OpenGL
MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Performing initial BA for initialization
MonoGS: Initialized SLAM
MonoGS: Evaluating ATE at frame:  113
Traceback (most recent call last):
  File "/home/marco/MonoGS/slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "/home/marco/MonoGS/slam.py", line 110, in __init__
    self.frontend.run()
  File "/home/marco/MonoGS/utils/slam_frontend.py", line 468, in run
    eval_ate(
  File "/home/marco/MonoGS/utils/eval_utils.py", line 105, in eval_ate
    ate = evaluate_evo(
  File "/home/marco/MonoGS/utils/eval_utils.py", line 29, in evaluate_evo
    traj_est_aligned = trajectory.align_trajectory(
AttributeError: module 'evo.core.trajectory' has no attribute 'align_trajectory'
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Resetting the opacity of non-visible Gaussians

What is the problem?

muskie82, 2024-04-01:
The error message says evo does not have align_trajector function.
Can you double check you have installed evo properly?

marcoastro, 2024-04-02:
> The error message says evo does not have align_trajector function. Can you double check you have installed evo properly?

Thanks for your answer！I had an error installing evo before, and it was fixed after reinstalling.

closed 2024-04-02 by marcoastro

Wingerllyyy, 2024-12-07:
Hello, I also encountered the same error message. I installed the same version of evo as the author or the latest version, but the error message still appeared. How did you reinstall it and which version did you install?