muskie82/MonoGS #54: Quick Demo Error

hnglp, 2024-04-02:
I have successfully downloaded the TUM dataset and added it to the correct file path. Additionally, the two modules mentioned in the warning at the beginning are installed within my environment. I don't believe there is any version mismatch issue either; I have checked each one. However, whenever I try to run the demo, I encounter the following error. If anyone has encountered a similar issue before and has a solution in mind, I would appreciate your assistance.

Best regards.

```
C:\Users\Unreal User\AppData\Roaming\Python\Python310\site-packages\torchvision\io\image.py:13: UserWarning: Failed to load image Python extension: '[WinError 127] The specified procedure could not be found'If you don't plan on using image functionality from `torchvision.io`, you can ignore this warning. Otherwise, there might be something wrong with your environment. Did you have `libjpeg` or `libpng` installed before building `torchvision` from source?
  warn(
MonoGS: saving results in results\datasets_tum\2024-04-02-17-41-13
Traceback (most recent call last):
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\slam.py", line 55, in __init__
    self.dataset = load_dataset(
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\utils\dataset.py", line 490, in load_dataset
    return TUMDataset(args, path, config)
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\utils\dataset.py", line 398, in __init__
    parser = TUMParser(dataset_path)
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\utils\dataset.py", line 51, in __init__
    self.load_poses(self.input_folder, frame_rate=32)
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\utils\dataset.py", line 86, in load_poses
    image_data = self.parse_list(image_list)
  File "C:\Users\Unreal User\Desktop\alp_akbaci_masterstudent\github_projects\MonoGS\utils\dataset.py", line 55, in parse_list
    data = np.loadtxt(filepath, delimiter=" ", dtype=np.unicode_, skiprows=skiprows)
  File "C:\Programms\anaconda3\envs\MonoGS\lib\site-packages\numpy\lib\npyio.py", line 1373, in loadtxt
    arr = _read(fname, dtype=dtype, comment=comment, delimiter=delimiter,
  File "C:\Programms\anaconda3\envs\MonoGS\lib\site-packages\numpy\lib\npyio.py", line 992, in _read
    fh = np.lib._datasource.open(fname, 'rt', encoding=encoding)
  File "C:\Programms\anaconda3\envs\MonoGS\lib\site-packages\numpy\lib\_datasource.py", line 193, in open
    return ds.open(path, mode, encoding=encoding, newline=newline)
  File "C:\Programms\anaconda3\envs\MonoGS\lib\site-packages\numpy\lib\_datasource.py", line 533, in open
    raise FileNotFoundError(f"{path} not found.")
FileNotFoundError: datasets/tum/rgbd_dataset_freiburg3_long_office_household\rgb.txt not found.
```

muskie82, 2024-04-03:
Hi,

Can you double check the downloaded dataset is on the correct path?
The error message says simply it can't find the dataset file on the path.

hnglp, 2024-04-03:
Alright, you're absolutely right. My files were in the correct location, but after extraction, they ended up in another directory within the main directory. That's why I encountered this error. We can close this issue now. Thank you.

closed 2024-04-03 by muskie82