muskie82/MonoGS #56: Apply to RGB monocular movie

RickMaruyama, 2024-04-03:
@muskie82 
Thank you very much for your great contribution to visual SLAM!!

I have a question.
How can your code apply to RGB monocular movie data?
I take a RGB monocular movie, and want to render it as ply format.

【detail】
I have reproduced the code on AWS EC2 (Ubuntu20.4 / A10 GPU) and successfully can get the GUI with demo dataset. (following code)
> python slam.py --config configs/mono/tum/fr3_office.yaml
Then I samely tried to use my movie data which is Monocular and produce RGB only. I inputted the RGB image and color info as rgb.txt and did same code. 
then the terminal requests me to give depth.txt.
> python slam.py --config configs/mono/tum/my_data.yaml
~~~~
  File "/home/ubuntu/MonoGS/utils/dataset.py", line 55, in parse_list
    data = np.loadtxt(filepath, delimiter=" ", dtype=np.unicode_, skiprows=skiprows)
  File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/npyio.py", line 1067, in loadtxt
    fh = np.lib._datasource.open(fname, 'rt', encoding=encoding)
  File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/_datasource.py", line 193, in open
    return ds.open(path, mode, encoding=encoding, newline=newline)
  File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/_datasource.py", line 533, in open
    raise IOError("%s not found." % path)
OSError: datasets/tum/rgb_dataset_chassy_crash_test/depth.txt not found.

I confirmed even depth.txt is added in the right folder and tried again the run slam.py, it still then need “pose_list”, probably related to accelerometer.txt and ground truth.txt.
How can I get them just with RGB camera? 

 python slam.py --config configs/mono/tum/my_data.yaml
MonoGS: saving results in 
results/tum_my_data/2024-04-03-06-50-39
Traceback (most recent call last):
  File "slam.py", line 251, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "slam.py", line 56, in __init__
    model_params, model_params.source_path, config=config
  File "/home/ubuntu/MonoGS/utils/dataset.py", line 490, in load_dataset
    return TUMDataset(args, path, config)
  File "/home/ubuntu/MonoGS/utils/dataset.py", line 398, in __init__
    parser = TUMParser(dataset_path)
  File "/home/ubuntu/MonoGS/utils/dataset.py", line 51, in __init__
    self.load_poses(self.input_folder, frame_rate=32)
  File "/home/ubuntu/MonoGS/utils/dataset.py", line 88, in load_poses
    pose_data = self.parse_list(pose_list, skiprows=1)
UnboundLocalError: local variable 'pose_list' referenced before assignment

So my question again is what is procedure to test on real data created by Monocular RGB camera? 
Thank you for your advice in advance.

rmurai0610, 2024-04-03:
To run our code with your custom data, you'll need to implement a dataset class
https://github.com/muskie82/MonoGS/blob/main/utils/dataset.py#L394-L402
which sets the following attributes (pseudocode):
```
self.has_depth = False
self.num_imgs = 100
self.color_paths = ["path/1.png", "path/2.png", ...]
self.depth_paths = None
self.poses = [np.eye(4) for _ in range(self.num_imgs)]
```
If you have gt poses, you can set the poses to be them.

RickMaruyama, 2024-04-09:
Thank you very much for your prompt reply and sorry for my late reply.
I successfully could see the rendering scene on GUI. however, somehow, I could not see the ply file output in result of gaussian splatting. I checked and got this info. Probably I could unset the depth value in wrong way..
could you advise me how to reach getting ply?
Thank you.
```

MonoGS/results/tum_rgb_dataset_sample_data/2024-04-04-03-19-38/plot/trj_final.json

{
    "trj_id": [
        80
    ],
    "trj_est": [
        [
            [
                1.0,
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                1.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                1.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0,
                1.0
            ]
        ]
    ],
    "trj_gt": [
        [
            [
                1.0,
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                1.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                1.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0,
                1.0
            ]
        ]
    ]
```

muskie82, 2024-04-15:
Hi,
[save_gaussians function](https://github.com/muskie82/MonoGS/blob/main/slam.py#L187C13-L187C27) exports ply file. 
To call this function, can you try one of the following options?
- Run with --eval flag.
- Set eval_rendering in config file True (like [this](https://github.com/muskie82/MonoGS/blob/main/configs/live/realsense.yaml#L7))
- Copy and paste save_gaussians function to where you want to call it.

Best,

RickMaruyama, 2024-04-16:
Thank you for your reply!

I have set eval_rendering = True, then Run with --eval.
to exclude any risk occurred from dataset, I use freriburg_desk rgb data as input. 
after running slam.py, I got asked to registered to wangb and did so. then I was guided to see the result in wagb, and no gui opened-up (gaussian-splatting) in my Ubuntu server.

I tired to investigate the reason why umeyama_alignment cannot work. it seems like the cause is groundtruth setting. Since I don't have groundtruth.txt in case of movie by iPhone(for example), the output of groudtruth could be wrong for your system.
How to treat groundtruth value (as well as accelerometer.txt if it's also must to set) in case I don't have it with monocular camera?

I hope this terminal output helps you to understand my problem..
```
**(MonoGS) ubuntu@ip-10-100-101-242:~/MonoGS$ python slam.py --config configs/mono/tum/dummy_data.yaml --eval**
MonoGS: Running MonoGS in Evaluation Mode
MonoGS: Following config will be overriden
MonoGS:         save_results=True
MonoGS:         use_gui=False
MonoGS:         eval_rendering=True
MonoGS:         use_wandb=True
MonoGS: saving results in results/tum_rgbd_dataset_freiburg1_desk copy/2024-04-16-03-31-58
wandb: Currently logged in as: rmaruyama7180 (use `wandb login --relogin` to force relogin)
wandb: wandb version 0.16.6 is available!  To upgrade, please run:
wandb:  $ pip install wandb --upgrade
wandb: Tracking run with wandb version 0.12.2
wandb: Syncing run configs/mono/tum/dummy_data_2024-04-16-03-31-58
wandb: ⭐️ View project at https://wandb.ai/rmaruyama7180/MonoGS
wandb:  View run at https://wandb.ai/rmaruyama7180/MonoGS/runs/eukpnf09
wandb: Run data is saved locally in /home/ubuntu/MonoGS/wandb/run-20240416_033159-eukpnf09
wandb: Run `wandb offline` to turn off syncing.

MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Performing initial BA for initialization
MonoGS: Initialized SLAM
MonoGS: Evaluating ATE at frame:  59

Traceback (most recent call last):
  File "slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "slam.py", line 110, in __init__
    self.frontend.run()
  File "/home/ubuntu/MonoGS/utils/slam_frontend.py", line 473, in run
    monocular=self.monocular,
  File "/home/ubuntu/MonoGS/utils/eval_utils.py", line 120, in eval_ate
    monocular=monocular,
  File "/home/ubuntu/MonoGS/utils/eval_utils.py", line 32, in evaluate_evo
    traj_est, traj_ref, correct_scale=monocular
  File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/evo/core/trajectory.py", line 397, in align_trajectory
    with_scale)
  File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/evo/core/geometry.py", line 72, in umeyama_alignment
    raise GeometryException("Degenerate covariance rank, "
**evo.core.geometry.GeometryException: Degenerate covariance rank, Umeyama alignment is not possible**

wandb: Waiting for W&B process to finish, PID 13615
wandb: Program failed with code 1.  Press ctrl-c to abort syncing.                                                                    
wandb: Find user logs for this run at: /home/ubuntu/MonoGS/wandb/run-20240416_033159-eukpnf09/logs/debug.log
wandb: Find internal logs for this run at: /home/ubuntu/MonoGS/wandb/run-20240416_033159-eukpnf09/logs/debug-internal.log
wandb: Synced 6 W&B file(s), 0 media file(s), 0 artifact file(s) and 0 other file(s)
wandb: Synced configs/mono/tum/dummy_data_2024-04-16-03-31-58: https://wandb.ai/rmaruyama7180/MonoGS/runs/eukpnf09
```

FYI. this is output to confirm with error-related variable value using python print func. It looks like pose_gt is cause of problem, letting y and mean_y to "0".
```
_pose_gt: [[1. 0. 0. 0.] [0. 1. 0. 0.] [0. 0. 1. 0.] [0. 0. 0. 1.]]
trj_gt_np: [array([[1., 0., 0., 0.],[0., 1., 0., 0.], [0., 0., 1., 0.], [0., 0., 0., 1.]]), 
array([[1., 0., 0., 0.], [0., 1., 0., 0.], [0., 0., 1., 0.], [0., 0., 0., 1.]]), 
 ...
array([[1., 0., 0., 0.], 0., 1., 0., 0.], 0., 0., 1., 0.], [0., 0., 0., 1.]])]
poses_gt: [array([[1., 0., 0., 0.], [0., 1., 0., 0.], [0., 0., 1., 0.], [0., 0., 0., 1.]]), 
 ....
array([[1., 0., 0., 0.],       [0., 1., 0., 0.],       [0., 0., 1., 0.],       [0., 0., 0., 1.]])]
 traj_ref = PosePath3D: 10 poses, 0.000m path length
n==-1でumeyama_alignment発動
traj_ref: 10 poses, 0.000m path length
y: [[0. 0. 0. 0. 0. 0. 0. 0. 0. 0.] [0. 0. 0. 0. 0. 0. 0. 0. 0. 0.] [0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]]
mean_y: [0. 0. 0.]
mean_x: [-0.00533411 -0.04845559 -0.25690131]
n: 10
outer_sum [[0. 0. 0.] [0. 0. 0.] [0. 0. 0.]]
cov_xy: [[0. 0. 0.] [0. 0. 0.] [0. 0. 0.]]
m-1 rank: 2
SVDv - d: [0. 0. 0.]
threshold - np.finfo(d.dtype).eps: 2.220446049250313e-16_
```

muskie82, 2024-04-18:
The easiest fix should be just to comment out eval_ate function.

https://github.com/muskie82/MonoGS/blob/37cebb4b8df24777980eac00276c07059b94762c/utils/slam_frontend.py#L349-L356

RickMaruyama, 2024-04-19:
Thanks for reply.
After comment out eval_ate at frontend.py, still same error about umeyama_ persists..

muskie82, 2024-04-19:
Of course here as well.
The idea is just to skip all the trajectory evaluation which requires gt pose.

https://github.com/muskie82/MonoGS/blob/37cebb4b8df24777980eac00276c07059b94762c/utils/slam_frontend.py#L468-L474

RickMaruyama, 2024-04-19:
@muskie82 
Yes, I've comment it out too.

muskie82, 2024-04-19:
OK, then where the umeyama alignment is called?
It is used just for trajectory evaluation, so just comment out all the evaluate_ate function in the code.

RickMaruyama, 2024-04-19:
it's here. 
/MonoGS/lib/python3.7/site-packages/evo/core/geometry.py
line 64, in umeyama_alignment

muskie82, 2024-04-19:
I mean, where in the MonoGS code it is called.

langsungn, 2024-05-15:
@RickMaruyama  Hi, can I ask how you succeeded in viewing the rendering GUI and implementing dataset.py? Thanks :)

RickMaruyama, 2024-05-15:
@langsungn 
Hello! I'm sorry for late feedback, I could get ply file but the plots clouds seems to be broken.
There maybe any issue in my modification..

langsungn, 2024-05-16:
@RickMaruyama Thank you for replying. I'm okay with a few issues, but would you share your modified 'dataset.py' file with me? Thanks a lot!

leichaoyang, 2024-05-21:
> To run our code with your custom data, you'll need to implement a dataset class https://github.com/muskie82/MonoGS/blob/main/utils/dataset.py#L394-L402 which sets the following attributes [...]

What does my_data.yaml format look like

leichaoyang, 2024-05-21:
> @muskie82 Thank you very much for your great contribution to visual SLAM!!
> 
> I have a question. How can your code apply to RGB monocular movie data? I take a RGB monocular movie, and want to render it as ply format.
> 
> 【detail】 I have reproduced the code on AWS EC2 (Ubuntu20.4 / A10 GPU) and successfully can get the GUI with demo dataset. (following code) `python slam.py --config configs/mono/tum/fr3_office.yaml` Then I samely tried to use my own movie data which is Monocular and produce RGB only. I inputted the RGB image and color info as rgb.txt and did same code. then the terminal requests me to give depth.txt.
> 
> ```
> python slam.py --config configs/mono/tum/my_data.yaml
> ~~~~
> File "/home/ubuntu/MonoGS/utils/dataset.py", line 55, in parse_list
> data = np.loadtxt(filepath, delimiter=" ", dtype=np.unicode_, skiprows=skiprows)
> File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/npyio.py", line 1067, in loadtxt
> fh = np.lib._datasource.open(fname, 'rt', encoding=encoding)
> File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/_datasource.py", line 193, in open
> return ds.open(path, mode, encoding=encoding, newline=newline)
> File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/_datasource.py", line 533, in open
> raise IOError("%s not found." % path)
> OSError: datasets/tum/rgb_dataset_chassy_crash_test/depth.txt not found.
> ```
> 
> I confirmed even depth.txt is added in the right folder and tried again the run slam.py, it still then need “pose_list”, probably related to accelerometer.txt and ground truth.txt. How can I get them just with RGB camera?
> 
> ```
> python slam.py --config configs/mono/tum/my_data.yaml
> MonoGS: saving results in 
> results/tum_my_data/2024-04-03-06-50-39
> Traceback (most recent call last):
>   File "slam.py", line 251, in <module>
>     slam = SLAM(config, save_dir=save_dir)
>   File "slam.py", line 56, in __init__
>     model_params, model_params.source_path, config=config
>   File "/home/ubuntu/MonoGS/utils/dataset.py", line 490, in load_dataset
>     return TUMDataset(args, path, config)
>   File "/home/ubuntu/MonoGS/utils/dataset.py", line 398, in __init__
>     parser = TUMParser(dataset_path)
>   File "/home/ubuntu/MonoGS/utils/dataset.py", line 51, in __init__
>     self.load_poses(self.input_folder, frame_rate=32)
>   File "/home/ubuntu/MonoGS/utils/dataset.py", line 88, in load_poses
>     pose_data = self.parse_list(pose_list, skiprows=1)
> UnboundLocalError: local variable 'pose_list' referenced before assignment
> ```
> 
> So my question again is what is procedure to test on real data created by Monocular RGB camera? Thank you for your advice in advance.

What does my_data.yaml format look like

leichaoyang, 2024-05-21:
> To run our code with your custom data, you'll need to implement a dataset class https://github.com/muskie82/MonoGS/blob/main/utils/dataset.py#L394-L402 which sets the following attributes [...]

class TUMDataset(MonocularDataset):
    def __init__(self, args, path, config):
        super().__init__(args, path, config)
        dataset_path = config["Dataset"]["dataset_path"]
        # parser = TUMParser(dataset_path)
        # self.num_imgs = parser.n_img
        # self.color_paths = parser.color_paths
        # self.depth_paths = parser.depth_paths
        # self.poses = parser.poses
        self.has_depth = False
        self.num_imgs = 82
        self.color_paths = None
        self.depth_paths = None
        self.poses = None
Modified the code. Run RGB images to display the following code：
Traceback (most recent call last):
  File "slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "slam.py", line 110, in __init__
    self.frontend.run()
  File "/data/ylc/MonoGS/utils/slam_frontend.py", line 375, in run
    self.dataset, cur_frame_idx, projection_matrix
  File "/data/ylc/MonoGS/utils/camera_utils.py", line 67, in init_from_dataset
    gt_color, gt_depth, gt_pose = dataset[idx]
  File "/data/ylc/MonoGS/utils/dataset.py", line 258, in __getitem__
    color_path = self.color_paths[idx]
TypeError: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/feishi/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/spawn.py", line 105, in spawn_main
    exitcode = _main(fd)
  File "/home/feishi/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/spawn.py", line 115, in _main
    self = reduction.pickle.load(from_parent)
  File "/home/feishi/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/synchronize.py", line 110, in __setstate__
    self._semlock = _multiprocessing.SemLock._rebuild(*state)
FileNotFoundError: [Errno 2] 没有那个文件或目录

leichaoyang, 2024-05-21:
@RickMaruyama Can you share your modified 'dataset.py' file with me？Thanks ！

RickMaruyama, 2024-05-22:
@leichaoyang 
I am sorry for my delay, since I was very busy for last days..
I attach dataset.py modified. Thank you for your support!

```

import csv
import glob
import os

import cv2
import numpy as np
import torch
import trimesh
from PIL import Image

from gaussian_splatting.utils.graphics_utils import focal2fov

try:
    import pyrealsense2 as rs
except Exception:
    pass

class ReplicaParser:
    def __init__(self, input_folder):
        self.input_folder = input_folder
        self.color_paths = sorted(glob.glob(f"{self.input_folder}/results/frame*.jpg"))
        #self.depth_paths = sorted(glob.glob(f"{self.input_folder}/results/depth*.png")) # RGBカメラ使用のためコメントアウト
        self.depth_paths = None # RGBカメラ使用のため追加
        self.n_img = len(self.color_paths)
        self.load_poses(f"{self.input_folder}/traj.txt")

    def load_poses(self, path):
        # self.poses = [] # RGBカメラ使用のためコメントアウト
        self.poses = [np.eye(4) for _ in range(self.n_img)]  # RGBカメラ使用のため追加
    
        with open(path, "r") as f:
            lines = f.readlines()

        frames = []
        for i in range(self.n_img):
            line = lines[i]
            pose = np.array(list(map(float, line.split()))).reshape(4, 4)
            pose = np.linalg.inv(pose)
            self.poses.append(pose)
            frame = {
                "file_path": self.color_paths[i],
                "depth_path": self.depth_paths[i],
                "transform_matrix": pose.tolist(),
            }

            frames.append(frame)
        self.frames = frames

class TUMParser:
    def __init__(self, input_folder):
        self.input_folder = input_folder
        self.load_poses(self.input_folder, frame_rate=32)
        self.n_img = len(self.color_paths)

    def parse_list(self, filepath, skiprows=0):
        data = np.loadtxt(filepath, delimiter=" ", dtype=np.unicode_, skiprows=skiprows)
        return data

    # def associate_frames(self, tstamp_image, tstamp_depth, tstamp_pose, max_dt=0.08):
    #     associations = []
    #     for i, t in enumerate(tstamp_image):
    #         if tstamp_pose is None:
    #             j = np.argmin(np.abs(tstamp_depth - t))
    #             if np.abs(tstamp_depth[j] - t) < max_dt:
[... 408 lines cut ...]
        image = cv2.cvtColor(image, cv2.COLOR_BGR2RGB)
        if self.disorted:
            image = cv2.remap(image, self.map1x, self.map1y, cv2.INTER_LINEAR)

        image = (
            torch.from_numpy(image / 255.0)
            .clamp(0.0, 1.0)
            .permute(2, 0, 1)
            .to(device=self.device, dtype=self.dtype)
        )
        return image, None, pose

def load_dataset(args, path, config):
    if config["Dataset"]["type"] == "tum":
        return TUMDataset(args, path, config)
    elif config["Dataset"]["type"] == "replica":
        return ReplicaDataset(args, path, config)
    elif config["Dataset"]["type"] == "euroc":
        return EurocDataset(args, path, config)
    elif config["Dataset"]["type"] == "realsense":
        return RealsenseDataset(args, path, config)
    else:
        raise ValueError("Unknown dataset type")

```

Harshitavardhani, 2024-05-24:
> @muskie82 Yes, I've comment it out too. Sorry for not to mention above.

Hi @muskie82 

Did you overcome the error regarding "Umeyama alignment is not possible"? I am facing the same issue while working with a custom dataset.

If so, could you please tell me how you resolved it? Thank you!

langsungn, 2024-05-25:
@leichaoyang Thank you soo much!! :)

RickMaruyama, 2024-05-31:
@langsungn 
Hello!  I'd like to solve this issue.
Do you have dataset.py for monocular camera dataset case, which successfully worked to get ply?
if so, I'd like to see it for my reference.

jdeng21, 2024-06-26:
@muskie82 @RickMaruyama @Harshitavardhani 
Hello! I am facing the same issue regarding "Degenerate covariance rank, Umeyama alignment is not possible". Did you solve the problem? If so, can you give me some advice? Thank you very much!

ShreyaIAB, 2024-08-13:
Hi I am getting the same error while using custom dataset. @Harshitavardhani @jdeng21 @langsungn were u able to solve it?

jdeng21, 2024-08-13:
@ShreyaIAB 
Hi! Try not run the eval_ate() function for the first few frames, because the trajectory might be 0 and that's why the error occurs.

ShreyaIAB, 2024-08-14:
@jdeng21, thank you. i commented out eval_ate(), and that did get rid of the **umeyama_alignment** error. my code does get stuck in the end. i am attaching the snap
[image]
For RGB, i was testing out with 35 RGB images. So my question is, along with eval_ate(), do i also need to make the modifications in dataset.py file as as posted in this thread?

ShreyaIAB, 2024-08-14:
@jdeng21 I also had one more question, I also want to test out with custom RGBD dataset. its the same 35 RGB images with their depth maps. I used the same trick, commented out eval_ate function, but i am getting a different error here in this case.
[image]
i have also changed the config file, where i have changed all the camera calibrations according to my iphone **except** depth_scale which is still at 5000.

MikeAiJF, 2024-09-05:
> @muskie82 非常感谢您对 SLAM 做出的巨大贡献！！
> 
> 我有问题。 您的代码如何获取RGB单眼电影数据？ 我拍摄了一部RGB单眼电影，并想将其渲染为ply格式。
> 
> 【详情】 我在AWS EC2（Ubuntu20.4 / A10 GPU）上删除了代码，并成功获得了带标签的演示数据集的GUI。（以下代码） `python slam.py --config configs/mono/tum/fr3_office.yaml` 然后我也尝试使用我自己的单目电影数据并仅产生RGB。我输入RGB图像和颜色信息作为rgb.txt并执行相同的代码。 然后终端要求我提供深度.txt。
> 
> ```
> python slam.py --config configs/mono/tum/my_data.yaml
> ~~~~
> File "/home/ubuntu/MonoGS/utils/dataset.py", line 55, in parse_list
> data = np.loadtxt(filepath, delimiter=" ", dtype=np.unicode_, skiprows=skiprows)
> File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/npyio.py", line 1067, in loadtxt
> fh = np.lib._datasource.open(fname, 'rt', encoding=encoding)
> File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/_datasource.py", line 193, in open
> return ds.open(path, mode, encoding=encoding, newline=newline)
> File "/home/ubuntu/miniconda3/envs/MonoGS/lib/python3.7/site-packages/numpy/lib/_datasource.py", line 533, in open
> raise IOError("%s not found." % path)
> OSError: datasets/tum/rgb_dataset_chassy_crash_test/depth.txt not found.
> ```
> 
> 我确认深度.txt已添加到正确的文件夹中，然后再次尝试运行slam.py，它仍然需要“pose_list”，可能与accelerometer.txt和groundtruth.txt有关。 我如何仅使用RGB相机获取它们？
> 
> ```
> python slam.py --config configs/mono/tum/my_data.yaml
> MonoGS: saving results in 
> results/tum_my_data/2024-04-03-06-50-39
> Traceback (most recent call last):
>   File "slam.py", line 251, in <module>
>     slam = SLAM(config, save_dir=save_dir)
>   File "slam.py", line 56, in __init__
>     model_params, model_params.source_path, config=config
>   File "/home/ubuntu/MonoGS/utils/dataset.py", line 490, in load_dataset
>     return TUMDataset(args, path, config)
>   File "/home/ubuntu/MonoGS/utils/dataset.py", line 398, in __init__
>     parser = TUMParser(dataset_path)
>   File "/home/ubuntu/MonoGS/utils/dataset.py", line 51, in __init__
>     self.load_poses(self.input_folder, frame_rate=32)
>   File "/home/ubuntu/MonoGS/utils/dataset.py", line 88, in load_poses
>     pose_data = self.parse_list(pose_list, skiprows=1)
> UnboundLocalError: local variable 'pose_list' referenced before assignment
> ```
> 
> 所以我的问题是，用单目 RGB 相机创建真实数据进行测试的程序是什么？ 提前感谢您的建议。
Hello, if I only have image data, can I do inference?