muskie82/MonoGS #59: torch.cuda.OutOfMemoryError

z157181773, 2024-04-05:
I encountered these two issues while running:
Question 1: [W CudaIPCTypes. cpp: 96] Producer process tried to deallocate over 1000 memory blocks referred by consumer processes Deallocation might be significantly slowed down We assume it will never go to be the case, but if it is, please file but to https://github.com/pytorch/pytorch
Problem 1 will not affect operation
Question 2: torch. cuda OutOfMemoryError: CUDA out of memory Tried to allocate 20.00 MiB GPU 0 has a total capacity of 7.75 GiB of which 13.31 MiB is free Process 4048 has 2.31 GiB memory in use Process 4130 has 278.00 MiB memory in use Including non PyTorch memory, this process has 5.09 GiB memory in use Of the allocated memory 4.64 GiB is allocated by PyTorch, and 315.91 MiB is reserved by PyTorch but unallocated If reserved but unallocated memory is large try setting PYTORCH-CUDA-ALLOCCONF=expandable segments: True to avoid fragmentation See documentation for Memory Management（ https://pytorch.org/docs/stable/notes/cuda.html#environment -Variables
The program will stop running
I would like to ask what is the reason for this? My GPU memory is 8188M

z157181773, 2024-04-05:
I saw it here https://github.com/muskie82/MonoGS/issues/31
What is the purpose of CUDA-LAUNCHBLOCKING=1 python slam. py -- config configs/mono/tum/fr3'office.yaml? I am using it very slowly now

z157181773, 2024-04-05:
Would it be better if I turn off the visual GUI?

closed 2024-04-09 by z157181773

Achleshwar, 2024-09-13:
Hi @rmurai0610 thanks for the great work!

I am also getting this warning:
`[W CudaIPCTypes. cpp: 95] Producer process tried to deallocate over 1000 memory blocks referred by consumer processes Deallocation might be significantly slowed down We assume it will never go to be the case, but if it is, please file but to https://github.com/pytorch/pytorch`

Do you have any suggestions on how to resolve this? I am trying to visualize trace plots to analyze the memory consumption during the process, but due to this warning, I am not able to move forward. TIA