muskie82/MonoGS #60: Question about rendering hole

Buffyqsf, 2024-04-09:
I got a hole in my own captured data as below, I've tried to find out the reason but failed.
[image]
And I found there is Gaussian, because when I change the view camera's z position, the hole dispeared like below.
Wondering if you have any experience in such situation.
[image]

Buffyqsf, 2024-04-09:
I found there seems to have a crop depth min(near 0.2), but I can't find any code can change this(I've tried to adjust near and far or fov in my view camera)

rmurai0610, 2024-04-09:
Can you try editing this line?
https://github.com/rmurai0610/diff-gaussian-rasterization-w-pose/blob/main/cuda_rasterizer/auxiliary.h#L154
You'll need to re-compile the diff-gaussian using
pip install -e submodules/diff-gaussian-rasterization

Buffyqsf, 2024-04-11:
well, it seems to be the line! Thank you very much! And I'll try it later.

closed 2024-04-11 by Buffyqsf

Buffyqsf, 2024-04-11:
I've tried but I still got the problem, it should be this line of code, but I don't know why it doesn't work. I've tried to bypass this by setting a unreal depth scale haha. Anyway thanks for your help!