muskie82/MonoGS #62: What exactly does this formula signify? It appears that it is not the equation for an ellipse, nor is it the equation for a circle.

chengYi-xun, 2024-04-09:
`float power = -0.5f * (conic.x * coordxy.x * coordxy.x + conic.z * coordxy.y * coordxy.y) - conic.y * coordxy.x * coordxy.y;`

chengYi-xun, 2024-04-09:
this code is in the gau_frag.glsl ,i know it is used to draw the  ellipse of guassian,but i can not know why use this equation

muskie82, 2024-04-18:
Hi,

It is the same exponential function as in [3DGS's rasterization pipeline](https://github.com/graphdeco-inria/diff-gaussian-rasterization/blob/59f5f77e3ddbac3ed9db93ec2cfe99ed6c5d121d/cuda_rasterizer/forward.cu#L335).
Further detail is in 7.1.2 of [EWA splatting paper](https://www.cs.umd.edu/~zwicker/publications/EWASplatting-TVCG02.pdf).

Best,

closed 2024-04-27 by muskie82