muskie82/MonoGS #65: RMSE ATE will suddenly become much larger

leqwq103, 2024-04-11:
[image]
Hello! Recently, while practicing on MonoGS code, I discovered an issue. During the operation, I,
MonoGS: Evaluating ATE at frame: 1025
Eval: RMSE ATE [m] 0.010714135020835631
MonoGS: Evaluating ATE at frame: 1121
Eval: RMSE ATE [m] 0.014316433431372504
MonoGS: Evaluating ATE at frame: 1200
Eval: RMSE ATE [m] 0.02489313051194339
MonoGS: Evaluating ATE at frame: 1286
Eval: RMSE ATE [m] 0.031591810183650405
RMSE ATE will suddenly become much larger. I checked the saved images and found that there is a curve in the map where the RMSE ATE value will increase. I would like to seek your help and see if there is a good solution?

leqwq103, 2024-04-11:
[image]
In the data set of rgbd_dataset_freiburg3_long_office_household, is the average RMSE ATE of MonoGS 0.0439m?
Similar to the final output:
Eval: RMSE ATE [m] 0.0439...

muskie82, 2024-04-15:
Hi,
It's understandable that some parts of the sequence, such as around the teddy bear in fr3/office due to strong rotations and close-ups, might show higher error.
As long as the final ATE RMSE number is reasonable, I think it is OK.

Best,

z157181773, 2024-04-15:
I also have a question about RMSE ATE because I am using Python slam.py -- config configs/mono/tum/fr3_office. yaml -- eval. Regarding mono/fr3 in the paper, the table in the paper states that RMSE ATE is 4.39cm. I would like to know if this is the average?After I improve the code, can I compare it with this value?

muskie82, 2024-04-15:
It is the average number of three runs as mentioned in the paper.
Yeah, just compare to it.

closed 2024-04-17 by leqwq103