muskie82/MonoGS #67: Question on Replica dataset

leblond14u, 2024-04-12:
Hi,

I'm trying to replicate your results and I want to test the replica dataset.
I have tried to load it with habitat-sim but I cannot seem to make it work. 

Could you tell me how you managed to obtain your replica data? 
And in case you used habitat-sim how to load it in the simulator.

Thanks a lot,
Best regards,

Hugo

muskie82, 2024-04-15:
Hi,

Since the dataset is introduced in iMAP paper and generated by the authors,  I am not sure about habitat-sim at all.

As a reference, vMAP repo has [habitat renderer code](https://github.com/kxhit/vMAP/blob/master/data_generation/habitat_renderer.py) and might be useful for you.

Best,

leblond14u, 2024-04-17:
Hi,

Thanks a lot, the vMAP code is indeed using habitat-sim to generate the replica scenes. 
Do you know by any chance how to get the Replica scene camera poses to generate the right path within replica ? 
The vMAP athors are using a `traj_w_c.txt` file that is not listed in the repository.

Best regards,

muskie82, 2024-04-18:
Hi,
I don't know the detail of trajectory generation process.

closed 2024-04-21 by muskie82