muskie82/MonoGS #71: RuntimeError: numel: integer multiplication overflow

zhongyingji, 2024-04-16:
Hi, thanks for the great work! 

I use your implemented package of diff-gaussian-rasterization, and I directly replace the rasterization package of original 3DGS with yours. 

Then I modify two parts as followings to input requirements of your code, 

[image]

[image]

And I got the error of "integer multiplication overflow": 

[image]

I tried to run with `CUDA-LAUNCHBLOCKING=1 python xxx", but the error kept the same. Do you have any idea on locating the error?

Thank you.

muskie82, 2024-04-30:
Thank you for your interest in our work.

Hmm... I never seen this before. Just to check, is the type of `viewpoint_camera.projection_matrix` torch.float32?

zhongyingji, 2024-05-06:
I solved in by https://github.com/graphdeco-inria/gaussian-splatting/issues/41. 

Thank you.

closed 2024-05-06 by zhongyingji