muskie82/MonoGS #76: Question about rasterizer

robofar, 2024-04-20:
Function `render` returns following dictionary:
```
 return {
        "render": rendered_image,
        "viewspace_points": screenspace_points,
        "visibility_filter": radii > 0,
        "radii": radii,
        "depth": depth,
        "opacity": opacity,
        "n_touched": n_touched,
    }
```

I could not figure out meaning of these keys in original and your rasterizer, so I would like to clarify. Could you explain me meaning of following keys, and what values stored in them represent. I know for each gaussian, value is stored for all keys, so could you explain me what these represent with respect to single gaussian and viewpoint?

1. `viewspace_points`
2. `radii`
3. `visibility_filter`
4. `n_touched`

Thank you very much!

rmurai0610, 2024-05-02:
Hi,
n_touched checks if the Gaussian was rasterised before we reach the median depth
https://github.com/rmurai0610/diff-gaussian-rasterization-w-pose/blob/main/cuda_rasterizer/forward.cu#L370
So even if the Gaussian is visible (in terms of radii) it may not be counted towards n_touched

robofar, 2024-05-05:
So, for n_touched, Gaussians whose depth < median_depth, they will be labeled with 1 in n_touched tensor.

For radii mask, all Gaussians in that are visible in viewpoint frustum will have value 1 in radii mask tensor?

rmurai0610, 2024-05-05:
n_touched counts it number of pixels which renders the Gaussian before the depth reaches median depth, so it will be greater than 0.
And yes, for the radii mask that's right. However, this only checks the radii of the projected Gaussian and does not take into account the occlusion like n_touched.