muskie82/MonoGS #91: Demo error w/wo GUI

hnglp, 2024-05-16:
Win 11
WSL/Ubuntu 20.04
Cuda 11.6

I successfully created the virtual environment, but when I tested the demo, I encountered the following error. Later, I tried without GUI using --evaluation and encountered the second error. I would be very happy if someone who can help could get back to me.

**Error 1 (with GUI)**

```
MonoGS: saving results in results/datasets_tum/2024-05-16-15-25-38
FEngine (64 bits) created at 0x561a479b5f30 (threading is enabled)
FEngine resolved backend: OpenGL
X Error of failed request:  GLXBadFBConfig
  Major opcode of failed request:  148 (GLX)
  Minor opcode of failed request:  0 ()
  Serial number of failed request:  27
  Current serial number in output stream:  27
MonoGS: Resetting the system
Process Process-3:
Traceback (most recent call last):
  File "/home/unreal_user/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/unreal_user/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/unreal_user/MonoGS/utils/slam_backend.py", line 406, in run
    self.initialize_map(cur_frame_idx, viewpoint)
  File "/home/unreal_user/MonoGS/utils/slam_backend.py", line 90, in initialize_map
    viewpoint, self.gaussians, self.pipeline_params, self.background
  File "/home/unreal_user/MonoGS/gaussian_splatting/gaussian_renderer/__init__.py", line 66, in render
    projmatrix=viewpoint_camera.full_proj_transform,
  File "/home/unreal_user/MonoGS/utils/camera_utils.py", line 102, in full_proj_transform
    self.projection_matrix.unsqueeze(0)
RuntimeError: CUDA error: CUBLAS_STATUS_NOT_SUPPORTED when calling `cublasSgemmStridedBatched( handle, opa, opb, m, n, k, &alpha, a, lda, stridea, b, ldb, strideb, &beta, c, ldc, stridec, num_batches)`
```

**Error 2 (without GUI)**

```
User
Traceback (most recent call last):
  File "/home/unreal_user/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/unreal_user/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/unreal_user/MonoGS/utils/slam_backend.py", line 417, in run
    self.add_next_kf(cur_frame_idx, viewpoint, depth_map=depth_map)
  File "/home/unreal_user/MonoGS/utils/slam_backend.py", line 69, in add_next_kf
    viewpoint, kf_id=frame_idx, init=init, scale=scale, depthmap=depth_map
  File "/home/unreal_user/MonoGS/gaussian_splatting/scene/gaussian_model.py", line 239, in extend_from_pcd_seq
    self.create_pcd_from_image(cam_info, init, scale=scale, depthmap=depthmap)
  File "/home/unreal_user/MonoGS/gaussian_splatting/scene/gaussian_model.py", line 131, in create_pcd_from_image
    return self.create_pcd_from_image_and_depth(cam, rgb, depth, init)
  File "/home/unreal_user/MonoGS/gaussian_splatting/scene/gaussian_model.py", line 150, in create_pcd_from_image_and_depth
    W2C = getWorld2View2(cam.R, cam.T).cpu().numpy()
  File "/home/unreal_user/MonoGS/gaussian_splatting/utils/graphics_utils.py", line 41, in getWorld2View2
    C2W = torch.linalg.inv(Rt)
torch._C._LinAlgError: torch.linalg.inv: The diagonal element 1 is zero, the inversion could not be completed because the input matrix is singular.
[W CudaIPCTypes.cpp:15] Producer process has been terminated before all shared CUDA tensors released. See Note [Sharing CUDA tensors]
```

closed 2024-08-29 by hnglp

bulecoder, 2024-12-20:
I also encountered problem two. How did you solve it

hnglp, 2024-12-20:
Now i use Linux instead of WSL and version 11.3 for CUDA