muskie82/MonoGS #92: No GUI pops up

ozyphus, 2024-05-19:
```
python slam.py --config configs/mono/tum/fr3_office.yaml 
MonoGS: saving results in results/datasets_tum/2024-05-19-20-22-26
cMonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Performing initial BA for initialization
MonoGS: Initialized SLAM
MonoGS: Evaluating ATE at frame:  111
Eval: RMSE ATE [m] 0.0064117039810880115
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Evaluating ATE at frame:  205
Eval: RMSE ATE [m] 0.015291334442466607
MonoGS: Evaluating ATE at frame:  296
Eval: RMSE ATE [m] 0.02375794441178842
MonoGS: Evaluating ATE at frame:  390
Eval: RMSE ATE [m] 0.0335763403092462
MonoGS: Evaluating ATE at frame:  507
Eval: RMSE ATE [m] 0.04069346619851069
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Evaluating ATE at frame:  607
Eval: RMSE ATE [m] 0.043461334584232326
MonoGS: Evaluating ATE at frame:  680
Eval: RMSE ATE [m] 0.042407737415241356
MonoGS: Evaluating ATE at frame:  778
Eval: RMSE ATE [m] 0.04125206880684835
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Evaluating ATE at frame:  897
Eval: RMSE ATE [m] 0.03990114477126345
MonoGS: Evaluating ATE at frame:  992
Eval: RMSE ATE [m] 0.038725210067905955
[W CudaIPCTypes.cpp:92] Producer process tried to deallocate over 1000 memory blocks referred by consumer processes. Deallocation might be significantly slowed down. We assume it will never going to be the case, but if it is, please file but to https://github.com/pytorch/pytorch
MonoGS: Evaluating ATE at frame:  1099
Eval: RMSE ATE [m] 0.038177517403356204

```

I don't see any GUI pop up. Does anyone have the same issue?

Harshitavardhani, 2024-05-28:
@ozyphus yes, we are getting the same issue but you will get point_clould.ply file. For you it will be in the results/datasets_tum/2024-05-19-20-22-26 path.

Ysc-shark, 2024-06-05:
> ```
> python slam.py --config configs/mono/tum/fr3_office.yaml
> MonoGS: saving results in results/datasets_tum/2024-05-19-20-22-26
> cMonoGS: Resetting the system
> MonoGS: Initialized map
> MonoGS: [...]

I get the same issue.

TracerChen, 2024-06-11:
I didn't have any GUI pop up either, but I can get the correct result by typing "xeyes" and so on.

ctl6998, 2024-07-03:
@Harshitavardhani how did you manage to get the .ply file? I run python slam.py --config configs/mono/tum/fr3_office.yaml  too and I only find .json and config.yml in the result