muskie82/MonoGS #99: Getting Trajectory file output and visualizing ply file

Mona-ShZeinoddin, 2024-05-31:
Hello and thank you for your amazing work. I wanted to ask how can I get the trajectory output file after the whole sequence? (the camera pose predicted by MonoGS, per  each frame, I'm looking for sth like a txt file that then I can evaluate against the groundtruth myself with evo (other than the evaluations that the model does))

Also, I wanted to ask about the ply file. I have tried visualizing the ply file in meshlab, but I lose color information this way (it just shows a set of points with the same color). When I use the website that you recommended (https://antimatter15.com/splat/) I get the splatted scene but still, I do not get any color info. I'm new to this so I'd be very grateful if you could guide me through this.

notu97, 2024-07-23:
The geometric depth is here i think: https://github.com/muskie82/MonoGS/blob/main/utils/slam_utils.py#L74

NagaPadma, 2024-09-12:
hello, have you tested it on custom dataset ? if yes, how did you create the groundtruth.txt ?