muskie82/MonoGS #102: GUI window didn't pop up

Ysc-shark, 2024-06-05:
When I run the Demo on a remote Ubuntu server, there are no error messages during the process, but the GUI window does not pop up. 

I tried entering 'xclock' to test, and the clock interface pops up normally. 

Why does this happen? How should I go about troubleshooting this issue?

TracerChen, 2024-06-11:
> When I run the Demo on a remote Ubuntu server, there are no error messages during the process, but the GUI window does not pop up.
> 
> I tried entering 'xclock' to test, and the clock interface pops up normally.
> 
> Why does this happen? How should I go about troubleshooting this issue?

Have you solved this problem?

qianqjia, 2024-06-23:
> Have you solved this problem?

I also have this issue.
Have you solved this problem?

ShreyaIAB, 2024-08-12:
export MESA_GL_VERSION_OVERRIDE=4.5, i solved the issue by typing that line before running the python script

tlxhlll, 2024-11-01:
I also have this issue

jiangguangan, 2024-11-12:
> export MESA_GL_VERSION_OVERRIDE=4.5，我通过在运行 Python 脚本之前键入该行解决了问题

[image]
What is the reason for the rendering result is not shown

CoderZhangYuana, 2025-03-09:
我也遇到了这个问题，这个问题可能产生的原因是你的系统 OpenGL 低于你的程序的 OpenGL 版本（例如 4.5），首先你可以使用
```
 glxinfo | grep "OpenGL version" 
```
 检查你的版本；如果你的系统上的 Mesa 版本较旧，可以尝试更新到最新版本。这可能会提供更好的 OpenGL 支持。
```
sudo apt update
sudo apt install mesa-utils
```
更改环境变量
```
export MESA_GL_VERSION_OVERRIDE=4.5
```
然后运行代码，发现GUI画面弹出，问题解决。

knightzzz9w, 2025-03-13:
> 我也遇到了这个问题，这个问题可能产生的原因是你的系统 OpenGL 低于你的程序的 OpenGL 版本（例如 4.5），首先你可以使用
>
> ```
>  glxinfo | grep "OpenGL version"
> ```
>
> 检查你的版本；如果你的系统上的 Mesa 版本较旧，可以尝试更新到最新版本。这可能会提供更好的 OpenGL 支持。
>
> ```
> sudo apt update
> sudo [...]

我在server上运行了
`glxinfo | grep OpenGL`
但是报错
`X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  66
  Current serial number in output stream:  67
`
why this happen