muskie82/MonoGS #104: Thank you for your interest in our work.

YIXIN-M, 2024-06-08:
Thank you for your interest in our work.

- To try rgb-d mode on dataset, just follow the readme.
The commands are 
```
bash scripts/download_tum.sh
python slam.py --config configs/rgbd/tum/fr3_office.yaml
```

- Our live demo with realsense in the video is done with monocular mode.
To try rgb-d mode on realsense live stream, need to change the [sensor type in config file](https://github.com/muskie82/MonoGS/blob/main/configs/live/realsense.yaml#L11) to "rgbd" and add few lines of code in [realsense data loader](https://github.com/muskie82/MonoGS/blob/main/utils/dataset.py#L469) to read depth images.

_Originally posted by @muskie82 in https://github.com/muskie82/MonoGS/issues/22#issuecomment-2000386413_

YIXIN-M, 2024-06-08:
Hello, what specific code do I need to add?

lordMutimir, 2024-10-09:
I got it working with this:
        self.has_depth = config["Dataset"]["sensor_type"] == "depth" or "rgb-d" 
In the RealsenseDataset class constructor in dataset.py.
And changing the realsense.yaml Dataset/sensor_type to 'rgb-d' was monocular.

lordMutimir, 2024-10-09:
NOTE: i am on the dev.speedup branch. And also i changed the timeout length in the get_item method in dataset.py:
        frameset = self.pipeline.wait_for_frames(20000) 
 Since the first frame sometimes takes more then 5 sec to appear.
But I am unsure if the IMU is used. I tried making a few gaussian splatting recordings and doing a loop in a room and the loop isn't closed perfectly. The offset was about a meter after a 4m circular walk in the room.