muskie82/MonoGS #105: Zero Error but Still Not Working

hnglp, 2024-06-10:
Hi,

I successfully completed the installation on Ubuntu 20.04 WSL without any errors. The GUI opens directly without any issues, but the map is not rendered correctly. I have attached an image below.

My graphics card is a GTX 3070. Could this be the cause of the problem?

I really need help thanks in advance!

[image]

MrVladimirLiu, 2024-07-08:
It seems wsl can't run this project correctly. I faced the same problem, the thread stuck in dead loop after Initialized map. I had build a docker image but it did't make sense, too.

MrVladimirLiu, 2024-07-08:
> It seems wsl can't run this project correctly. I faced the same problem, the thread stuck in dead loop after Initialized map. I had build a docker image but it did't make sense, too.

I copy the same image to a linux server and it works well.

hnglp, 2024-07-22:
[image]
[image]

After a while ı got these results but after this i have a new error about matrix singularity again. Any idea of this situation?

hnglp, 2024-08-29:
I changed my operating system from Windows to Linux and my problems were solved. Pytorch works with some problems on Windows

closed 2024-08-29 by hnglp

FraDePace, 2024-12-09:
@hnglp Hi, were you then able to run it with WSL?