muskie82/MonoGS #107: TypeError in slam_gui.py during GUI usage in MonoGS

thuThink, 2024-06-13:
Hi,

I am encountering an issue while running MonoGS with the GUI. The GUI freezes, and the following error traceback is shown:

```bash
MonoGS: Evaluating ATE at frame:  446
Eval: RMSE ATE [m] 0.013367331140081758
MonoGS: Resetting the opacity of non-visible Gaussians
Process Process-4:
Traceback (most recent call last):
  File "/home/huan/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/huan/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/huan/project/MonoGS/gui/slam_gui.py", line 688, in run
    app.run()
  File "/home/huan/project/MonoGS/gui/slam_gui.py", line 676, in update
    self.scene_update()
  File "/home/huan/project/MonoGS/gui/slam_gui.py", line 662, in scene_update
    self.receive_data(self.q_main2vis)
  File "/home/huan/project/MonoGS/gui/slam_gui.py", line 394, in receive_data
    gaussian_packet = get_latest_queue(q)
  File "/home/huan/project/MonoGS/gui/gui_utils.py", line 148, in get_latest_queue
    message_latest = q.get_nowait()
  File "/home/huan/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/queues.py", line 126, in get_nowait
    return self.get(False)
  File "/home/huan/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/queues.py", line 113, in get
    return _ForkingPickler.loads(res)
  File "/home/huan/anaconda3/envs/MonoGS/lib/python3.7/site-packages/torch/multiprocessing/reductions.py", line 129, in rebuild_cuda_tensor
    tensor_offset, tensor_size, tensor_stride)
  File "/home/huan/anaconda3/envs/MonoGS/lib/python3.7/site-packages/torch/_utils.py", line 134, in _rebuild_tensor
    return t.set_(storage._untyped(), storage_offset, size, stride)
TypeError: set_() received an invalid combination of arguments - got (torch.storage._UntypedStorage, int, Tensor, tuple), but expected one of:
 * ()
 * (torch.Storage source)
 * (torch.Storage source, int storage_offset, tuple of ints size, tuple of ints stride)
      didn't match because some of the arguments have invalid types: (torch.storage._UntypedStorage, int, Tensor, tuple)
 * (Tensor source)
 * (Tensor source, int storage_offset, tuple of ints size, tuple of ints stride)
      didn't match because some of the arguments have invalid types: (torch.storage._UntypedStorage, int, Tensor, tuple)

MonoGS: Evaluating ATE at frame:  520
Eval: RMSE ATE [m] 0.014609354080900642

```
Any guidance on how to resolve this issue would be greatly appreciated.

Thank you!

JIAZHAOQIAN, 2024-07-05:
I modified the function named 'get_latest_queue' in gui_utils.py as follows:
<pre>
def get_latest_queue(q):
    message = None
    while True:
        try:
            message_latest = q.get_nowait()
            if message is not None:
                del message
            message = message_latest
        except queue.Empty:
            if q.qsize() < 1:
                break
        # zajia: for unsolved bug related to "torch.storage._UntypedStorage"
        except TypeError:
            print("get a torch.storage._UntypedStorage error!")
            break
    return message
</pre>
I try to capture the error and do not modifiy the 'message' variable. It seems work properly.

zuoweic, 2024-08-15:
I met save issue, but why?

Wingerllyyy, 2024-10-13:
I also met this. Have you found out the reason?