muskie82/MonoGS #111: Cannot Install Conda Virtual Env

czeyveli1, 2024-07-07:
Hello everyone;
I try to install virtual env using `conda env create -f environment.yml` but I take this error:
```
`Channels:
 - pytorch
 - conda-forge
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages:

Preparing transaction: done
Verifying transaction: done
Executing transaction: | By downloading and using the CUDA Toolkit conda packages, you accept the terms and conditions of the CUDA End User License Agreement (EULA): https://docs.nvidia.com/cuda/eula/index.html

done
Installing pip dependencies: | Ran pip subprocess with arguments:
['/home/cz/anaconda3/envs/MonoGS/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/cz/Documents/MonoGS/condaenv.l32swvkn.requirements.txt', '--exists-action=b']
Pip subprocess output:
Processing ./submodules/simple-knn
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'

Pip subprocess error:
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      Traceback (most recent call last):
        File "<string>", line 36, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/home/cz/Documents/MonoGS/submodules/simple-knn/setup.py", line 30, in <module>
          extra_compile_args={"nvcc": [], "cxx": cxx_compiler_flags})
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 983, in CUDAExtension
          library_dirs += library_paths(cuda=True)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1098, in library_paths
          if (not os.path.exists(_join_cuda_home(lib_dir)) and
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 2125, in _join_cuda_home
          raise EnvironmentError('CUDA_HOME environment variable is not set. '
      OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

failed

CondaEnvException: Pip failed
`
```
Could you help me that how can I solve this problem. Also I used to "--recursive"  command cloning this repo.

ctl6998, 2024-07-08:
conda install -c conda-forge cudatoolkit-dev

You must set up the CUDA_HOME, this work in my case:

Ref: https://github.com/conda/conda/issues/7757

ShreyaIAB, 2024-07-10:
i have the same issue

harshal-14, 2024-07-12:
Did you solve this issue? I'm having a similar issue

notu97, 2024-07-23:
Follow this: https://github.com/muskie82/MonoGS/issues/114#issuecomment-2244199923

czeyveli1, 2024-07-27:
@notu97 this in also not work. After applied your command, when I try to install other packages, I get an error: 
```
Pip subprocess error:
ERROR: Could not find a version that satisfies the requirement open3d (from versions: none)
ERROR: No matching distribution found for open3d'
```

notu97, 2024-07-27:
yes, so install open3d
`pip install open3d==0.17.0`

czeyveli1, 2024-07-28:
I solved the problem, and I will try to explain how I did that:

1) I tried to set up this package using the normal installation guide:
```
conda env create -f environment.yml
conda activate MonoGS
```
However, I still encountered the same error.

2) I activated the virtual environment and installed Python 3.10 (`conda install python=3.10`) because Python 3.7 does not support some libraries.

3) I tried to install specific versions of PyTorch, Torchvision, Torchaudio, and CUDA using:
`conda install pytorch==2.3.1 torchvision==0.18.1 torchaudio==2.3.1 pytorch-cuda=12.1 -c pytorch -c nvidia`

4) I installed other packages manually using:
`pip install submodules/simple-knn submodules/diff-gaussian-rasterization opencv-python==4.8.1.78 munch trimesh evo==1.11.0 open3d torchmetrics imgviz PyOpenGL glfw PyGLM wandb lpips rich ruff`

notu97, 2024-07-28:
> I solved the problem, and I will try to explain how I did that:
>
> 1. I tried to set up this package using the normal installation guide:
>
> ```
> conda env create -f environment.yml
> conda activate MonoGS [...]

while running MonoGS you might encounter problem from the evo libraray. Just add `ax=ax` in the plot_trajectory call inside the trajectory.py file in the evo library.

closed 2024-07-29 by czeyveli1

hadhuba, 2024-09-22:
Hi I tried your solution but unfortunately it didn't work for me. If there is any additional advice you have I would be happy to hear them. (I am desperate)

I believe, that I have set the CUDA_HOME properly, and cloned the repo with the flag --recursive. 
However I am having challanges with installing simple_knn, and diff_gaussian_rasterization. The rest of the environment.yml file is perfectly installed,  and the following are working as expected as well:
import torch
print(torch.cuda.is_available()) # This should return true.
print(torch.cuda.get_device_name(0))  # You should see the name of

Thanks in advance