muskie82/MonoGS #112: combine with vins or orbslam3

knightzzz9w, 2024-07-10:
The paper mentioned that you add camera pose from orbslam3. I have tried a similar way, add vins pose estimation result in MonoGS, and then don't update the pose. The ATE error is smaller , but I can't get a better rendering result compared to baseline. And I have tried to update keyframe pose in the backend, but still get a lower psnr(in MH02 , baseline psnr 27, my result is 23). I wonder to know why there is a decrease in the PSNR, and how you add orbslam3's pose into the whole system?

Thanks!

minsoo5403, 2024-07-15:
I tried the same approach, but in MonoGS, fixing the pose to the GroundTruth Pose without rendering updates affects accuracy.

In the internal algorithm, the camera parameters are optimized together with the pose after rendering, so fixing them to the GT values leads to an inaccurate map update.

xixishui1999, 2025-02-10:
@knightzzz9w Hello, I would like to ask how to add VINS pose estimation results in MonoGS.
Thanks!