muskie82/MonoGS #113: 其他数据序列表现不佳问题

kongxiangxi026514, 2024-07-10:
I tested using other data sequences on the TUM dataset, such as fr2upioneer_160, and did not change the number of iterations. However, I found that in situations with larger scene sizes, there was a significant localization error. May I ask why this is?

kongxiangxi026514, 2024-08-15:
> Are you using mono mode or depth mode? If mono, it does have a poor value in some larger reconstruction scene due to losing scale. If depth, try to check the quality of depth image, the TUM RGB-D are taken in real world, the quality of depth image is catastrophically poor, much blank and noisy, it will significantly disturb the joint loss of geometric loss.

Thanks for the reply, I'm using the depth mode, I also think it's a depth inaccurate issue, is there any good way to check the depth image, please?

UltraHertzz, 2024-08-22:
try to check the difference of trj change in both mono and depth mode. usually the mono mode should be stable but lost its scale, and the depth mode will suddenly drift into none space in some key points, so lang as i test it. Find out these timestamp and check the quality of depth, such as how many zero or NaN value in it