muskie82/MonoGS #114: Conda environment setup issue

harshal-14, 2024-07-11:
Hi I'm not able to setup the conda environment with the help of environment.yaml file. Kindly help me debug this issue:

```
(base) lucifer@lucifer-G3-3500:~/WPI/Summer_Internship/MonoGS$ conda env create -f environment.yml
Channels:
 - pytorch
 - conda-forge
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages:
                                                                                                                                                                                                            
Preparing transaction: done                                                                                                                                                                                 
Verifying transaction: done                                                                                                                                                                                 
Executing transaction: \ By downloading and using the CUDA Toolkit conda packages, you accept the terms and conditions of the CUDA End User License Agreement (EULA): https://docs.nvidia.com/cuda/eula/index.html                                                                                                                                                                                                      
                                                                                                                                                                                                            
done                                                                                                                                                                                                        
Installing pip dependencies: - Ran pip subprocess with arguments:                                                                                                                                           
['/home/lucifer/miniconda3/envs/MonoGS/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/lucifer/WPI/Summer_Internship/MonoGS/condaenv.q20019ia.requirements.txt', '--exists-action=b']               
Pip subprocess output:                                                                                                                                                                                      
Processing ./submodules/simple-knn                                                                                                                                                                          
  Preparing metadata (setup.py): started                                                                                                                                                                    
  Preparing metadata (setup.py): finished with status 'done'                                                                                                                                                
Processing ./submodules/diff-gaussian-rasterization                                                                                                                                                         
  Preparing metadata (setup.py): started                                                                                                                                                                    
  Preparing metadata (setup.py): finished with status 'done'                                                                                                                                                
Collecting opencv-python==4.8.1.78                                                                                                                                                                          
  Downloading opencv_python-4.8.1.78-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (61.7 MB)                                                                                                     
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.7/61.7 MB 2.0 MB/s eta 0:00:00                                                                                                                             
Collecting munch                                                                                                                                                                                            
  Downloading munch-4.0.0-py2.py3-none-any.whl (9.9 kB)                                                                                                                                                     
Collecting trimesh                                                                                                                                                                                          
  Downloading trimesh-4.4.1-py3-none-any.whl (694 kB)                                                                                                                                                       
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 694.7/694.7 kB 1.5 MB/s eta 0:00:00                                                                                                                           
Collecting evo==1.11.0                                                                                                                                                                                      
  Downloading evo-1.11.0.tar.gz (113 kB)                                                                                                                                                                    
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 114.0/114.0 kB 2.2 MB/s eta 0:00:00                                                                                                                           
  Preparing metadata (setup.py): started                                                                                                                                                                    
  Preparing metadata (setup.py): finished with status 'done'                                                                                                                                                
Collecting open3d==0.17.0                                                                                                                                                                                   
  Downloading open3d-0.17.0-cp37-cp37m-manylinux_2_27_x86_64.whl (420.4 MB)                                                                                                                                 
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 420.4/420.4 MB 1.5 MB/s eta 0:00:00                                                                                                                           
Collecting torchmetrics                                                                                                                                                                                     
  Downloading torchmetrics-0.11.4-py3-none-any.whl (519 kB)                                                                                                                                                 
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 519.2/519.2 kB 1.8 MB/s eta 0:00:00                                                                                                                           
Collecting imgviz                                                                                                                                                                                           
  Downloading imgviz-1.7.5.tar.gz (7.7 MB)                                                                                                                                                                  
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.7/7.7 MB 2.1 MB/s eta 0:00:00                                                                                                                               
  Preparing metadata (setup.py): started                                                                                                                                                                    
  Preparing metadata (setup.py): finished with status 'done'
Collecting PyOpenGL
[... cut ...]
      PyTorch (11.6). Please make sure to use the same CUDA versions.
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> simple_knn

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
WARNING: There was an error checking the latest version of pip.

failed

CondaEnvException: Pip failed
```

notu97, 2024-07-23:
Donot use the environment.yaml file to prepare the conda environment directly. Follow this process step by step:
- Install NVIDIA GPU Drivers: (see website and download run file and run it.
- Do `nvidia-smi` on a terminal to verify installation. 
- Install conda and create a environment
- Activate the conda env.
- Install cuda-nvcc inside the conda environment: https://anaconda.org/nvidia/cuda-nvcc
        -conda install nvidia/label/cuda-12.1.0::cuda-nvcc
- Check cuda version by "nvcc --version", you should see cuda 12.1
- Install cudatoolkit: https://anaconda.org/nvidia/cuda-toolkit
        -conda install nvidia/label/cuda-12.1.0::cuda-toolkit 
- Go to Pytorch website and install the stable version of pytorch
        -conda install pytorch torchvision torchaudio pytorch-cuda=12.1 -c pytorch -c nvidia
- Verify installation:
```
import torch
print(torch.cuda.is_available()) # This should return true.
print(torch.cuda.get_device_name(0))  # You should see the name of your GPU
```
- Next install all the other pending packages one my one from the envirnment.yaml

czeyveli1, 2024-07-27:
Could you explain how did you install other pending packages one by one? Because when I try to install the simple-knn package, I get an error.

notu97, 2024-07-27:
What is the error that you are getting. Here is the command to install simple-knn (run from the MonoGS directory
`pip install submodules/simple-knn`

czeyveli1, 2024-07-27:
```
  Running setup.py install for simple-knn ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for simple-knn did not run successfully.
  │ exit code: 1
  ╰─> [56 lines of output]
      No CUDA runtime is found, using CUDA_HOME='/home/cz/anaconda3/envs/MonoGS'
      running install
      /home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        setuptools.SetuptoolsDeprecationWarning,
      running build
      running build_ext
      /home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/torch/utils/cpp_extension.py:476: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      Traceback (most recent call last):
        File "<string>", line 36, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/home/cz/Documents/MonoGS/submodules/simple-knn/setup.py", line 33, in <module>
          'build_ext': BuildExtension
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/__init__.py", line 87, in setup
          return distutils.core.setup(**attrs)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/dist.py", line 1208, in run_command
          super().run_command(command)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/command/install.py", line 68, in run
          return orig.install.run(self)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/command/install.py", line 698, in run
          self.run_command('build')
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/dist.py", line 1208, in run_command
          super().run_command(command)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/command/build.py", line 132, in run
          self.run_command(cmd_name)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/dist.py", line 1208, in run_command
          super().run_command(command)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 84, in run
          _build_ext.run(self)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 346, in run
          self.build_extensions()
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 499, in build_extensions
          _check_cuda_version(compiler_name, compiler_version)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 382, in _check_cuda_version
          torch_cuda_version = packaging.version.parse(torch.version.cuda)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/pkg_resources/_vendor/packaging/version.py", line 49, in parse
          return Version(version)
        File "/home/cz/anaconda3/envs/MonoGS/lib/python3.7/site-packages/pkg_resources/_vendor/packaging/version.py", line 264, in __init__
          match = self._regex.search(version)
      TypeError: expected string or bytes-like object
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> simple-knn

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

czeyveli1, 2024-07-27:
I think the problem is about the version of CUDA. My first CUDA version was 12.1 but it didnt work. I have changed the CUDA version to 11.7 but get same error...

notu97, 2024-07-28:
didi you check `nvcc --version` inside your Conda environment?

czeyveli1, 2024-07-28:
```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Tue_May__3_18:49:52_PDT_2022
Cuda compilation tools, release 11.7, V11.7.64
Build cuda_11.7.r11.7/compiler.31294372_0
```

notu97, 2024-07-28:
[image]
install cuda 12.1 and its corresponding tool kit. then use the command in above image to install the correct pytorch version.
or install the correct pytorch version for cuda 11.7:
`conda install pytorch==2.0.1 torchvision==0.15.2 torchaudio==2.0.2 pytorch-cuda=11.7 -c pytorch -c nvidia`

czeyveli1, 2024-07-28:
When I check the NVIDIA GPU drivers using `nvidia-smi` command, my Nvidia Cuda version is` NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.2`, but we still try to install the 12.1 version. Does it cause my problem?

notu97, 2024-07-28:
> When I check the NVIDIA GPU drivers using `nvidia-smi` command, my Nvidia Cuda version is` NVIDIA-SMI 535.183.01 Driver Version: 535.183.01 CUDA Version: 12.2`, but we still try to install the 12.1 version. Does it cause my problem?

No, that doesn't cause any problem.

1425562478, 2024-11-29:
I met a same problem like you, how did you do with this problem?thanks