muskie82/MonoGS #116: Live demo with Realsense

Tanfuchy, 2024-07-17:
(MonoGS)$ python slam.py --config configs/live/realsense_rgbd.yaml
Traceback (most recent call last):
  File "slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "slam.py", line 56, in __init__
    model_params, model_params.source_path, config=config
  File "/home/chy/mycode/MonoGS/utils/dataset.py", line 530, in load_dataset
    return RealsenseDataset(args, path, config)
  File "/home/chy/mycode/MonoGS/utils/dataset.py", line 432, in __init__
    self.pipeline = rs.pipeline()
NameError: name 'rs' is not defined

 python slam.py --config configs/live/realsense.yaml
Traceback (most recent call last):
  File "slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "slam.py", line 56, in __init__
    model_params, model_params.source_path, config=config
  File "/home/chy/mycode/MonoGS/utils/dataset.py", line 530, in load_dataset
    return RealsenseDataset(args, path, config)
  File "/home/chy/mycode/MonoGS/utils/dataset.py", line 432, in __init__
    self.pipeline = rs.pipeline()
NameError: name 'rs' is not defined

harshal-14, 2024-08-06:
I guess you are missing the pyrealsense library installation. Ensure you have that installed in your environment:
https://pypi.org/project/pyrealsense2/

Make sure you import it at the top of the file.

MikeAiJF, 2024-09-04:
[image]
How should this error be resolved

lordMutimir, 2024-10-09:
Follow this:
https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md
If you manage to see it using the realsense-viewer you should not get this error.