muskie82/MonoGS #117: CUDA error: CUBLAS_STATUS_INVALID_VALUE when calling `cublasSgemmStridedBatched

azuryl, 2024-07-21:
python slam.py --config configs/mono/tum/fr3_office.yaml
MonoGS: saving results in results/datasets_tum/2024-07-21-03-56-53
MonoGS: Resetting the system
Process Process-3:
Traceback (most recent call last):
  File "/home/azuryl/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/azuryl/anaconda3/envs/MonoGS/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/data/azuryl/MonoGS/utils/slam_backend.py", line 406, in run
    self.initialize_map(cur_frame_idx, viewpoint)
  File "/data/azuryl/MonoGS/utils/slam_backend.py", line 90, in initialize_map
    viewpoint, self.gaussians, self.pipeline_params, self.background
  File "/data/azuryl/MonoGS/gaussian_splatting/gaussian_renderer/__init__.py", line 66, in render
    projmatrix=viewpoint_camera.full_proj_transform,
  File "/data/azuryl/MonoGS/utils/camera_utils.py", line 102, in full_proj_transform
    self.projection_matrix.unsqueeze(0)
RuntimeError: CUDA error: CUBLAS_STATUS_INVALID_VALUE when calling `cublasSgemmStridedBatched( handle, opa, opb, m, n, k, &alpha, a, lda, stridea, b, ldb, strideb, &beta, c, ldc, stridec, num_batches)`
FEngine (64 bits) created at 0x545c0080 (threading is enabled)
FEngine resolved backend: OpenGL

xytjc, 2024-11-12:
i have the same problem,did you solve it?

xytjc, 2024-11-12:
@azuryl bro，i have solve it by :unset LD_LIBRARY_PATH
more detiles in 
https://blog.csdn.net/BetrayFree/article/details/133868929

kc12ww, 2025-04-30:
unset LD_LIBRARY_PATH