muskie82/MonoGS #118: Saving the entire 3D gaussian scene.

notu97, 2024-07-24:
Hi, is there any way to store the entire 3D gaussian splatted scene with the color and everything (similar to the one that is shown on the GUI when the program runs). I see that there is a ply file, but turns out it is just point cloud, there is no RGB values in it. Thanks.

TommasoLabieni, 2024-08-01:
You can visualize the gaussians by dragging that PLY file here: https://antimatter15.com/splat/

ShreyaIAB, 2024-08-06:
is there a .obj format file generated? is there an option?

UltraHertzz, 2024-08-14:
or use Simple splatting visualizer [here](https://github.com/limacv/GaussianSplattingViewer), open ply file and change max_sh to 0 in gui_util.py in it. I have write a adaptive sh value visualizer that can read from config, i will raise a pull request recently.

UltraHertzz, 2024-08-14:
> is there a .obj format file generated? is there an option?

Probably not, the .obj will store vertices and other mesh like structure, so far as i know there is no feature_dc, feature_rest and sh in .obj format, or you can have a try to convert it into mesh, some mature work has been raised up.