muskie82/MonoGS #125: ImportError: Failed to import any qt binding

Niamh5163, 2024-08-08:
Hi, I have the following problem when trying to run the command python slam.py --config configs/mono/tum/fr3_office.yaml and would like to ask how to fix it? Thank you very much for your help！

`MonoGS: saving results in results/datasets_tum/2024-08-08-11-08-53
FEngine (64 bits) created at 0x385d3280 (threading is enabled)
FEngine resolved backend: OpenGL
MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Resetting the opacity of non-visible Gaussians
MonoGS: Performing initial BA for initialization
MonoGS: Initialized SLAM
MonoGS: Evaluating ATE at frame:  107
Eval: RMSE ATE [m] 0.007149343724647586
Traceback (most recent call last):
  File "slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "slam.py", line 110, in __init__
    self.frontend.run()
  File "/home/yjj/MonoGS/utils/slam_frontend.py", line 473, in run
    monocular=self.monocular,
  File "/home/yjj/MonoGS/utils/eval_utils.py", line 110, in eval_ate
    monocular=monocular,
  File "/home/yjj/MonoGS/utils/eval_utils.py", line 50, in evaluate_evo
    fig = plt.figure()
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 809, in figure
    FigureClass=FigureClass, **kwargs)
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 324, in new_figure_manager
    _warn_if_gui_out_of_main_thread()
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 314, in _warn_if_gui_out_of_main_thread
    if (_get_required_interactive_framework(_get_backend_mod())
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 217, in _get_backend_mod
    switch_backend(dict.__getitem__(rcParams, "backend"))
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 282, in switch_backend
    class backend_mod(matplotlib.backend_bases._Backend):
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 283, in backend_mod
    locals().update(vars(importlib.import_module(backend_name)))
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/backends/backend_qt5agg.py", line 7, in <module>
    from .backend_qtagg import (    # noqa: F401, E402 # pylint: disable=W0611
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/backends/backend_qtagg.py", line 9, in <module>
    from .qt_compat import QT_API, _enum, _setDevicePixelRatio
  File "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/backends/qt_compat.py", line 142, in <module>
    raise ImportError("Failed to import any qt binding")
ImportError: Failed to import any qt binding
MonoGS: Resetting the opacity of non-visible Gaussians
`

Niamh5163, 2024-08-08:
After the last problem, I installed PyQt5 and then the runtime problem became this. Can anyone help me see what the problem is, much appreciated!

QObject::moveToThread: Current thread (0xc2c6a290) is not the object's thread (0xb5aa1270).
Cannot move to target thread (0xc2c6a290)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/yjj/anaconda3/envs/MonoGS/lib/python3.7/site-packages/cv2/qt/plugins". even though it was found.
This application failed to start because no Qt platform plugin could be initialised. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland- xcomposite-glx, webgl.

Abandoned (core dumped)

zuoweic, 2024-08-15:
My GUI works normally for a few seconds and shows same problems as your first post ...

zuoweic, 2024-08-15:
My situation is exactly same, i meet your first problem then i installed PyQt5 then  i meet the same problem..........

zuoweic, 2024-08-15:
Dude, i solved it!!!!!!
pip install opencv-python==4.1.1.26
pip install opencv-contrib-python==4.1.1.26
It works for me

Niamh5163, 2024-08-15:
Thank you soooooo much！

Niamh5163, 2024-08-15:
> Dude, i solved it!!!!!! pip install opencv-python==4.1.1.26 pip install opencv-contrib-python==4.1.1.26 It works for me

Thank you soooooo much！

closed 2024-08-15 by Niamh5163

JzHuai0108, 2024-08-28:
Make sure pip has installed PyQt5, and then take a look at [this issue](https://github.com/muskie82/MonoGS/issues/131). In my case, I installed opencv-python-headless 4.10, and opencv-contrib-python 4.10.