muskie82/MonoGS #126: Wandb error

ShreyaIAB, 2024-08-09:
MonoGS: saving results in results/datasets_tum/2024-08-09-09-08-55
Traceback (most recent call last):
  File "slam.py", line 249, in <module>
    wandb.define_metric("frame_idx")
  File "/home/sinha/.conda/envs/MonoGS/lib/python3.7/site-packages/wandb/sdk/wandb_run.py", line 401, in wrapper_fn
    return func(self, *args, **kwargs)
  File "/home/sinha/.conda/envs/MonoGS/lib/python3.7/site-packages/wandb/sdk/wandb_run.py", line 391, in wrapper
    return func(self, *args, **kwargs)
  File "/home/sinha/.conda/envs/MonoGS/lib/python3.7/site-packages/wandb/sdk/wandb_run.py", line 2750, in define_metric
    overwrite,
  File "/home/sinha/.conda/envs/MonoGS/lib/python3.7/site-packages/wandb/sdk/wandb_run.py", line 2816, in _define_metric
    tel.feature.metric = True
  File "/home/sinha/.conda/envs/MonoGS/lib/python3.7/site-packages/wandb/sdk/lib/telemetry.py", line 42, in __exit__
    self._run._telemetry_callback(self._obj)
  File "/home/sinha/.conda/envs/MonoGS/lib/python3.7/site-packages/wandb/sdk/wandb_run.py", line 799, in _telemetry_callback
    self._telemetry_obj.MergeFrom(telem_obj)
AttributeError: 'Run' object has no attribute '_telemetry_obj'

taehun-ryu, 2024-08-09:
I solved the same error using a lower version of the wandb. I use `0.16.6' version.
```pip install wandb==0.16.6```
I hope it works for you.

ShreyaIAB, 2024-08-09:
Thank you, i will check it out